InfiniTimeOrg/InfiniTime #548: Wake on wrist raise too sensitive

drzacek, 2021-08-02:
I think the wake-up on wrist raise is too sensitive. 

While it is nice to not have to twist your arm quickly by 90 degrees to activate the screen (like on my MiBand 4), it is pointles to have it activated almost constantly during normal day activities. 

My PineTime lights up even when I'm driving on a highway and barely turn the steering wheel. 

I think that battery life suffer greatly from that. Maybe tuning it down a little bit might be a good idea.

Dperrigo18, 2021-08-02:
I would recommend something like a sensitivity slider in the settings for this would be good.
That way the user could fine tune it to what would be acceptable for them.

atomdmac, 2021-08-03:
https://github.com/JF002/InfiniTime/issues/490

hubmartin, 2021-08-03:
I've did some experiments with Colmi P8 accelerometer filtering because it was also turning on too often. I solved that by averaging last accelerometer 4 values. https://github.com/hubmartin/InfiniTime/blob/64758b76247ccbb25c5ee84786b60d34618aa277/src/drivers/Bma421.cpp#L126

Today the PineTime arrived so I did a test and PineTime BMA accelerometer code is really too much sensitive. Comparison video.
https://photos.app.goo.gl/pEP5nXSFnpNkJMQcA

We can solve that by different finite state machines and "AI" but the simple average filter works actually good. It might turn on when you raise wrist with something in your hand, but it will not turn on when watches are flat and you bump into something.
I noticed that in the BMA code there is some averaging filter in hardware. Not sure how it works or if it will need to increase that a bit.

drzacek, 2021-08-03:
Averaging the last values does sound good. 
I only got mine last week and I'm still setting my build env. Once I got it working I'm gonna play around with stuff.

Riksu9000, 2021-08-03:
>I have also 100Hz sampling and 2g range (reading over I2C seems like 10 Hz over my scope)

The sensor is polled at 10Hz in systemTask. This should be improved by enabling interrupts from the sensor, and reading the values into a small buffer. The frequency might not have to be much higher than 10Hz.

>We can solve that by different finite state machines and "AI" but the simple average filter works actually good.

There's no need to make this complicated. The current wakeup code is very basic. With a small buffer and a bit higher update frequency it could easily be improved by requiring some velocity rather than looking at just the angle.

https://github.com/JF002/InfiniTime/blob/514481ef7f9c71ad816b31d979c6ab39ce9380dd/src/components/motion/MotionController.cpp#L12

PR #826 (2023-08-17): Improve Raise to Wake algorithm
Depends on #1696 and #1726.

Fixes #548.

With this implementation, your hand needs to be raised at a certain speed to register.

The values for the speed and threshold are values that feel about right to me, but I'm up for changing them.
closed 2023-08-17 by JF002