InfiniTimeOrg/InfiniTime #559: Make color definitions more global in source code

drzacek, 2021-08-05:
I'm not sure this is the right place to bring such stuff, if not then sorry. 

I've been trying to change the colors for the notification app today (not really a fan of the original), and I noticed that there are many defines such `LV_COLOR_ORANGE` in many places, or sometimes even `lv_color_hex(0x888888)`. So if you want to change colors for whole screen, you need to find all occurrences and replace them with new color. 

I think if we had some global defines in separate header file, with more user-(programmer-)friendly names, like LV_COLOR_BG_CLOCKAPP, LV_COLOR_TEXT_NOTIFICATIONTITLE or something, with LV_COLOR_RED/BLUE/something assigned to it, managing themes would become way easier. 
Creators of new watchfaces/themes would just have to work with single .h file and it all would become a bit more readable.

hubmartin, 2021-08-05:
I can imagine that these colors will be part of some "theme" defined in RAM or LittleFS filesystem config file which could be easily changed on watch or over BLE.
Thinking further - there could be one watchface which will be dynamic and will be loaded from LittleFS storage. I'm not talking about scripting, but simple definition of text type, format, colors , font size and position. This way you wouldn't need to reflash firmware for different watchfaces.
I'm working on BLE text console and one of the ideas it could be used is changing parameters of different objects on the screen. Since everything is in RAM, you could theoretically change anything on the screen on the fly. Just list LVGL objects on the screen and redefine font, position, color..

leroivi, 2025-03-01:
Hello,

I was playing with colors on my custom watchface and I didn't know where to put my colors and the gradient function. In fact the colors are still splitted in the source code between 3 places : The Infinitheme, the PinetimeStyle and the lvgl library.
I would like to suggest a refactor that centralize colors in one place that expose the available color palette, then the themes can refer to this palette, so does the screens.