InfiniTimeOrg/InfiniTime #572: [Feature Request] NFC support

raulgool, 2021-08-12:
I noticed that the microprocessor used in watch has NFC tag module as well.
I was wondering, in future, if it will be possible to implement contact-less payment like google-pay or apple-pay?
I am not really sure about the requirements for implementing similar functionality.

If not payment, I am guessing hardware can still be programmed as a NFC key-fob.

Avamander, 2021-08-12:
> I noticed that the microprocessor used in watch has NFC tag module as well.

Nope, no antenna.

> I was wondering, in future, if it will be possible to implement contact-less payment like google-pay or apple-pay?

Nope, no hardware security functionality that would allow such certification.

raulgool, 2021-08-12:
> Nope, no antenna.

I see. I thought it was there because in the schematic on pin 30 of nRF52832 it shows some antenna connection.

moriel5, 2021-08-12:
If I am not mistaken, there is a conversation about this in the Pine64 forums (in the PineTime subforum, of course).

I think it was said there, that you could connect an NFC antennae and have the module work, but due to how the PineTime it's wired, the screen won't be able to go to low power modes.

closed 2021-08-12 by raulgool