InfiniTimeOrg/InfiniTime #573: Question: Debugging w/ DevContainer

atomdmac, 2021-08-12:
**TLDR**: My questions are:
* Am I correct in my assumption that I can push new software to the watch and debug it from within the VS Code dev container?
* Are there any additional steps (such as manually starting the OpenOCD server on my host machine?) that need to be taken to facilitate this.

**Background:**

With the recent addition of the VSCode dev container, I've finally been able to get some code compiled and sent to my dev watch using the OTA method.  This is all well and good but:
* The process is slow and makes iterating on changes arduous
* It affords no debugging capabilities (ex. logging, etc.)

I've read through the various docs that describe using OpenOCD for flashing the application firmware to the watch as well as (I think?) connecting to it for debugging purposes.  It looks like debugging facilities are set up within the dev container as well (accessed via the `Run and Debug` tab in VS Code).

From reading the various docs, it appears that the dev container provides facilities for flashing the application firmware to the watch as well as debugging it.  (Granted, I could be mistaken about this).

However, when attempting to run + debug, VS Code presents me with a pop-up message that reads: 

`Failed to launch GDB: host.docker.internal:3333: Connection timed out. (from target-select extended-remote host.docker.internal:3333)`

hubmartin, 2021-08-12:
Hi, what OS and container backend do you use? Docker/VMWare/WSL2?
@geekbozu helped me so I know a little bit now :)

In Ubuntu I had to run OpenOCD locally (outside of docker) and do this https://github.com/JF002/InfiniTime/issues/569 to connect from the docker. Other OS might be different.

Or after compilation in docker, you can choose my debug configuration which runs openocd automatically (Ubuntu) and expects precise filename of the *.out file. I do not use bootloader, so I cannot help you with bootloader combo.
https://github.com/JF002/InfiniTime/pull/567

atomdmac, 2021-08-13:
Hi @hubmartin !  I'm using MacOS (Mojave) with Docker.

For the debug configuration, does OpenOCD run automatically directly on the _host_?  Or within the dev container?

hubmartin, 2021-08-13:
Openocd is not run inside the container, because it cannot dig the usb inside. So you run it on your host system with port 3333. Then start debug and in the `launch.json` there is the first one configuration which should connect to your host port 3333.

But you can append another launch config from #567 and this one runs openocd automatically. Once you have compiled sources and created binaries in docker, they are in your host filesystem, so you could run my launch config. You just need to configure the path to arm-none-eabi-gdb which now has to installed somewhere in your host system.