InfiniTimeOrg/InfiniTime #576: PineTime always reboots to "INFINITIME" logo and gets stuck there, until firmware is reflashed

unilock, 2021-08-13:
My (sealed!) PineTime is running firmware v1.3.0, and is connected to my Google Pixel 4a via GadgetBridge v0.59.0.

Sometimes I need to reboot my watch because it loses connection with GadgetBridge, as discussed in #302.

Recently, after no identifiable trigger, every reboot leads to the "INFINITIME" logo instead of the clockface. Whether I'm rebooting from a functional OS or from the InfiniTime logo itself makes no difference.

I'm able to get back to the OS after reflashing the latest firmware ZIP, but rebooting even immediately afterwards leads to the stuck InfiniTime logo again.

I'm not sure how to diagnose this issue, as my PineTime is sealed, and I don't have the equipment to diagnose it anyway.

unilock, 2021-08-13:
Validating the firmware after reflashing it seems to have fixed the issue. I guess the firmware corrupted itself?

closed 2021-08-13 by unilock

Itai-Nelken, 2021-08-13:
If you don't validate the firmware, when reseting the bootloader will revert to the last used firmware, in your case the recovery firmware.

Debo-rah, 2021-08-17:
Hi ,

I have the same problem here on a sealed PineTIme but I am sure I did validate the firmware. What can I do? How can i reflash the firmware when it is in this "infinitme logo mode"?
Thanks for help!

Itai-Nelken, 2021-08-17:
@Debo-rah the "InfiniTime logo mode" isbthe recovery firmware. Connect to it using nrfconnect (only app I know that works with it) and dfu the firmware.

Debo-rah, 2021-08-17:
Ah, thanks a lot! I only used GadgetBridge so far and it coudnt connect. I will try nrf connect later.

tlhonmey, 2023-08-30:
I've had this happen twice now.  On two different watches on two different firmware revisions.  Firmware was verified both times.  First time my two-year-old was playing with it, and the second time I put the watch in my pocket for a few minutes to avoid damaging it.

For anyone else finding this, the second time the watch was still connected to gadget bridge and would vibrate whenever a notification went by (though obviously not display it).  If it's still doing that then gadgetbridge seems to work just fine to reflash it.  Can't speak to any of the other options or if it's not already talking.

Anybody know if this is actually a feature where you're supposed to be able to kick it back to the recovery firmware on purpose?

And the weird part on this second one is that it seems to have killed the accelerometer completely.  Not even detected.

JF002, 2023-09-02:
The InfiniTime logo is actually the recovery firmware. It's automatically loaded by the bootloader if it cannot find any firmware that runs correctly. It can also be manually loaded by keeping the button pushed when rebooting (until the pine cone is drawn in red). See [this](https://github.com/InfiniTimeOrg/pinetime-mcuboot-bootloader#boot-flow) for more info about the boot process.

When the recovery firmware is loaded, you can reboot into the normal firmware simply by rebooting (keep the button pressed until it reboots). 

However, if you load the recovery multiple times consecutively, this will erase the normal firmware, and the only firmware available to the boot loader will be the recovery one. This can happen if you keep the button pushed across multiple reboots... which can happen when the watch is stored in a pocket, a bag or given to a young toddler :)

To work around this, simply reinstall infinitime by re-uploading it using your companion app. And don't forget to validate it so the bootloader loads InfiniTime instead of the recovery firmware next time the watch reboots!

sammyj85, 2023-12-14:
> And the weird part on this second one is that it seems to have killed the accelerometer completely. Not even detected.

Can you please explain that more?

I'm in a situation where I didn't even have any recovery firmware problems (just using the watch as normal), and my accelerometer stopped functioning completely. There is no more wake on [action], and calibrate shake does not register anything. At the same time, my friend found his watch with the same issue - both on firmware 1.13.0.

It may be some hidden configuration state that stops the accelerometer working, as he reverted his firmware to 1.11.0, and the accelerometer started working again.

sammyj85, 2023-12-14:
> To work around this, simply reinstall infinitime by re-uploading it using your companion app. And don't forget to validate it so the bootloader loads InfiniTime instead of the recovery firmware next time the watch reboots!

It may well be worth noting that not all companion apps are equal. Siglo and Watchmate (last time I tried) both don't work when the watch is in recovery firmware mode.

minacode, 2023-12-14:
> Siglo and Watchmate (last time I tried) both don't work when the watch is in recovery firmware mode.

Not sure how active the development of Siglo is, but you should open an issue for Watchmate for this.