InfiniTimeOrg/InfiniTime #578: Cover screen to sleep

Arsen6331, 2021-08-14:
Apple's WatchOS has a very nice feature where you can cover the screen with your hand, and the watch goes to sleep. Can this be added to InfiniTime? Is it even possible on the PineTime?

hubmartin, 2021-08-14:
Not an expert on touch controller, but what I read it can return up to 10 touch points and it also sends size of every touch point. So maybe if the hand is reported like 1 point with really big size, we can trigger an event to do something.

Riksu9000, 2021-08-14:
The touch controller can only detect a single touch, and can't detect pressure. Some of the documentation is misleading. The PineTIme can't detect when the screen is covered.

JF002, 2021-08-14:
At the beginning of the project, I think someone said the touch controller was great and supported multi-touch. However, I've never managed to detect more than one touch point, as @Riksu9000 said. I don't see any way to implement this feature with the sensors we currently have on board, unfortunately.

darkdragon-001, 2021-10-03:
Phones usually use an (optical) proximity sensor to detect such events. PineTime doesn't have one on board, so I agree that it doesn't work well with the current sensors...

Have a look on #713 which might solve your need (sleep when lowering hand).

geekbozu, 2021-10-04:
Indeed, the Touch controller is very poorly documented. And it may support some of these extra features but as of the moment there is no implementation for them. Nor any clear direction on how to enable or support them. Will take some RE work to implement if at all possible.

wmertens, 2022-09-16:
FYI the mi band also has this feature, from 4 at least, so maybe the touch controller can detect it

JF002, 2022-09-17:
I'll close this feature request as the feature is not documented and (probably) not supported by the touch controller.

closed 2022-09-17 by JF002