InfiniTimeOrg/InfiniTime #583: Firmware freezing after doubletap to sleep

Riksu9000, 2021-08-15:
Swipe from QuickSettings to watchface and doubletap during the animation to go to sleep. Now the watch won't wake up and will reset soon.

I could replicate the issue in 1.2.0, but not 1.1.0. Its harder to do in 1.2.0 than 1.3.0.

On 1.2.0 it sometimes just stays at a black screen and it must be reset by holding the button. I didn't get this in 1.3.0.

Can someone confirm this isn't just my watch and some register I may have changed in the touch panel for example that hasn't reset?

https://user-images.githubusercontent.com/37774658/129474007-143e46b6-cf05-4e0f-94e6-6a95fede7fcd.mp4

lman0, 2021-08-15:
@Riksu9000 what firmware version do you use ? 
because in 1.3.0 i can't reproduce it , since double tap can only awake but not sleep the pinetime

Riksu9000, 2021-08-15:
This started happening somewhere in between 1.1.0 and 1.2.0 and is still present in latest develop for me.

lman0, 2021-08-15:
maye it related to the twin things that reset when an refresh is ongoing

hubmartin, 2021-08-15:
Can confirm I was able to have same reset behaviour with 1.3.0.
(i do not use the analog clock, also never knew I could go back to sleep with doubleclick :) )

JF002, 2021-08-18:
> (i do not use the analog clock, also never knew I could go back to sleep with doubleclick :) ) 

This is not supposed to happen :-D

Avamander, 2021-10-08:
If this is still an issue, let's reopen.

Riksu9000, 2021-10-08:
It is. I think it was after #568 that it doesn't freeze every time, but can cause screen corruption instead.

AndreySV, 2022-08-01:
still exists in 1.10.0

minacode, 2022-12-15:
Still exists in 1.11 as well.  
I could not reproduce it in any other setup like the Infineat or the Casio watch face which both also take a little moment to show.

mark9064, 2024-12-11:
Can anyone still repro this?

minacode, 2024-12-11:
I cannot reproduce it on 1.15.

mark9064, 2024-12-17:
Going to close for now, let's re-open this if the problem returns

closed 2024-12-17 by mark9064