InfiniTimeOrg/InfiniTime #588: InfiniTime 1.3.0 Ukrainian characters ignores

AndriiZ, 2021-08-16:
InfiniTime 1.3.0 ignores Ukrainian characters (і,ї,є)  when showing notifications (other Cyrillic characters displaying fine)

Avamander, 2021-08-17:
Have your companion app transliterate while  waiting for larger fonts to be included at some point.

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/212

closed 2021-10-02 by Avamander

ITCactus, 2021-12-23:
@AndriiZ  i prepared the "fix" for that in PR #887 with support for Ukrainian language.
unfortunately it will not be in mainline/upstream, but you can download my repo, and make a build for yourself.
eventually, i will try to make a release in my fork, but as for now, only DEBUG releases are flashed OK for me, but RELEASE are not (maybe it's relates to some latest changes in the upstream... i will try again after 1.8 release).