InfiniTimeOrg/InfiniTime #589: Is there a reason why the numbers arent centered in PineTimeStyle?

HolySoap, 2021-08-17:
Is this space on the left reserved for more informations or is it an oversight?

[image]

Riksu9000, 2021-08-17:
When using 12 hour format it shows AM/PM on the left side. It could possibly be improved still.

HolySoap, 2021-08-17:
Alright, makes sense thanks for the answer.
Moving the am/pm to the right underneath the battery would be the best solution I would think.

kieranc, 2021-08-18:
It was actually intentional, and happened before the AM/PM icon was placed... I just felt it looked better a little bit to the right!

HolySoap, 2021-08-18:
> It was actually intentional, and happened before the AM/PM icon was placed... I just felt it looked better a little bit to the right!

Thats a great example of how different people look at it differently.
That it isn't in the middle bothers me so much that I get triggered the moment I see this Watch Face.

JF002, 2021-09-02:
I have to agree I haven't noticed that the number were not centered until I read this issue.
I'm closing this issue as the original question have been answered.

closed 2021-09-02 by JF002