InfiniTimeOrg/InfiniTime #590: Unresponsive touchscreen: no swiping / taping | no heartbeat monitor

MonsieurPaulLeBoulanger, 2021-08-17:
#Description

I have just received my PineTime yesterday and the touchscreen doesn't respond : I can't swipe (any way) so I'm unable to use the watch.
The step counter works properly but my heartbeat is not detected.
I can sync the watch with my laptop (siglo) and my smartphone (gadetbridge).

I tried to update the firmware using siglo but I can't validate it as I'm unable to swipe to the firmware screen.
I also tried a hard reset (draining the battery).

#What's expected?
Swiping on the touch screen should show other screens with icons, buttons...
The heartbeat monitor should detect my heartbeat as, afaik, I'm not dead ;)

#Environment
    PineTime sealed
    InfiniTime 1.0.0
    Bootloader 0.14.1
    Recovery 0.14.1

Riksu9000, 2021-08-17:
If the touchscreen never responds, this sounds like a hardware fault. You should probably contact Pine64 to get a replacement.

MonsieurPaulLeBoulanger, 2021-08-17:
Sure, I was wondering if I could do more tests & checks, just in case.

Avamander, 2021-08-17:
If it's an unsealed version, just reset the touch panel cable first.

Riksu9000, 2021-08-17:
You could try flashing this. It tries to read the Chip id, vendor id and firmware version from the touch controller and display it in the digital watch face. I'm assuming you can only see the digital watch face. If this shows ff or freezes, we can be certain that the touch controller isn't working.

[image]

[pinetime-mcuboot-app-dfu-1.3.0.zip](https://github.com/JF002/InfiniTime/files/7000019/pinetime-mcuboot-app-dfu-1.3.0.zip)

MonsieurPaulLeBoulanger, 2021-08-17:
hmm, I guess `0 0 0` isn't really expected.
[image]

Riksu9000, 2021-08-17:
Interesting. It seems that communication with the chip was successful, but maybe the chip is partially faulty?

MonsieurPaulLeBoulanger, 2021-08-17:
@Riksu9000 let me know if I can try something else.
And thank you for your help ;)

MonsieurPaulLeBoulanger, 2021-09-09:
Just received a spare PineTime that works fine.

closed 2021-09-09 by MonsieurPaulLeBoulanger

asdfzerozero, 2021-12-02:
I seem to be having this issue - I do not use this often but I tried to upgrade to 1.7.1 and I am still frozen.  unsure how to troubleshoot at this point -- wondering what list of tools I can use to debug? (closed pinetime)