InfiniTimeOrg/InfiniTime #591: Bug: Vibrating during a call

drzacek, 2021-08-17:
Pinetime was displaying the notification with telephone buttons when I called somebody, kept vibrating for the whole duration of the call. 

Now it is debatable, if it has to display it at all for outgoing calls, but i lt surely shouldn't vibrate during a call - once it is started, it should stop.

Riksu9000, 2021-08-18:
It doesn't display a notification on outgoing call for me. Are you using the latest Gadgetbridge?

The notification can be dismissed without pressing any of the buttons by swiping up on develop branch currently. This will also stop the vibration.

However it seems that the notification doesn't automatically get dismissed when the call is accepted or rejected on the phone, which is a bug.

drzacek, 2021-08-18:
Well it happened to me when pinetime was charging on the table, and I was unable to get to it during the call.

JF002, 2021-08-18:
@drzacek Which companion app do you use? I've never noticed that behavior with Gadgetbridge on Android. Also, as call notifications were modified in #544, can you tell us which version of Infinitime you were using (released version or build from develop)?

> However it seems that the notification doesn't automatically get dismissed when the call is accepted or rejected on the phone, which is a bug.

@Riksu9000 I don't think the companion app has anyone to dismiss a notification, unfortunately. I've just tested with the debug call notification from Gadgetbridge, and indeed, since #544, the call notification vibrates until you swipe up or push the button to dismiss the notification. Maybe the vibration should stop after some time (I don't know, 5 or 10 vibrations)? What do you think?

Riksu9000, 2021-08-18:
There should be some timeout that stops it eventually, but are you sure it can't be done through Gadgetbridge? There are start and end call notifications in the debug page of Gadgetbridge, and I'd assume start call means that the call was picked up in this case.

[image]

hubmartin, 2021-08-18:
I can confirm the same behaviour on my Pine with develop branch. I received a call and after a few minutes during the call I looked at my watch on table and there was incomng call and watch was vibrating shortly every second.

drzacek, 2021-08-19:
I am using gadgetbridge, version 0.59.0.
The Infinitime is based on develop branch from few days ago, once I'm near my pc I can check when exactly I pulled.

Got incoming call today, pt was vibrating for the whole duration of the call AND didn't stopped after the call was over. Had to manually dismiss it (by clicking red button).

JF002, 2021-08-19:
> There should be some timeout that stops it eventually, but are you sure it can't be done through Gadgetbridge? There are start and end call notifications in the debug page of Gadgetbridge, and I'd assume start call means that the call was picked up in this case.

Yep, Gadetbridge seems to provide that functionality, but I meant that the protocol we use for notifications (the Alert Notification Service, ANS) do not provide any way to dismiss notifications once they are sent (but I should check the spec to the sure, though).

Riksu9000, 2021-08-19:
I think this issue stems from the fact that call notifications are considered similar to other notifications, when they probably shouldn't.

The call screen with buttons should probably be entirely separate from Notifications. When a call starts, this screen pops up, but no notification is pushed to the Notifications list yet. When a start or end call alert is received or a button is pressed, it will simply close this screen. At that point it could push a simple text notification to the notifications that says "call with xxx ended" or something similar.

geekbozu, 2021-08-21:
> 
> 
> I think this issue stems from the fact that call notifications are considered similar to other notifications, when they probably shouldn't.
> 
> The call screen with buttons should probably be entirely separate from Notifications. When a call starts, this screen pops up, but no notification is pushed to the Notifications list yet. When a start or end call alert is received or a button is pressed, it will simply close this screen. At that point it could push a simple text notification to Notifications that says "call with xxx ended" or something similar.

I agree, Calls should definitely be more of an "APP" then a notification, we could then extend it to be a proper call app as well to make/place calls if desired in the future.

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/367

closed 2021-10-02 by Avamander