InfiniTimeOrg/InfiniTime #592: Feature request: Notify on fully charged

drzacek, 2021-08-17:
I like the feature of my Miband, that vibrates and displays info when it has reached full battery. 

I think it would be nice to have this too on pinetime. 

I'm not a huge fan of the battery level symbol, I'd rather have percentage value displayed, but since we only have the icon, it is hard to tell if it's fully charged or it still has to go for 10 more percent. Having the notitication would solve that.

geekbozu, 2021-09-03:
I would like this as well actually, However we need to actually get battery stats working first...right now its not the most consistent at knowing when its actually charged across devices due to the way we measure the battery level.

Riksu9000, 2021-09-03:
> I would like this as well actually, However we need to actually get battery stats working first...right now its not the most consistent at knowing when its actually charged across devices due to the way we measure the battery level.

The battery is full when power present is true and charging is false. I'm testing something that uses this right now.