InfiniTimeOrg/InfiniTime #594: Bug: Heart rate monitoring continues while charging

FrazerClews, 2021-08-17:
If the heart rate monitor is on before charging, it will continue to stay on during and after charging.

I have made a WIP branch [here](https://github.com/JF002/InfiniTime/compare/develop...FrazerClews:frazer/heartReadingStopp) for now before I go any further with the patch which needs testing, just in case something else is planned to fix this.

Riksu9000, 2021-08-18:
It also keeps on measuring if it's just placed on a table. The ambient light sensor should probably be used instead to detect when the watch is taken off and then the measurement should be stopped.

hatmajster, 2021-08-18:
Erm, its not true, monitoring is turned off as soon as the screen go black, exactly like on hand - so its not working all the time while charging.