InfiniTimeOrg/InfiniTime #597: Feature request - vibration on bluetooth disconnect

AndriiZ, 2021-08-18:
It would be great to get a notification that  you lost your phone.
Vibration should be 1-2 sec to be noticeable

Riksu9000, 2021-08-18:
Disconnection isn't the same as losing your phone. Just going into another room is enough to disconnect and not everyone keeps their phones with them always.

AndriiZ, 2021-08-18:
> 
> 
> Disconnection isn't the same as losing your phone. Just going into another room is enough to disconnect and not everyone keeps their phones with them always.

I know. And for me it is good that my watch notices me about disconnect (current behavior of Honor Band 4). At least, this may be option disabled by default

hamblingreen, 2021-09-07:
duplicate of #410

adocampo, 2021-09-14:
> Disconnection isn't the same as losing your phone. Just going into another room is enough to disconnect and not everyone keeps their phones with them always.

I think that's why pebble notification showed "Where is my phone". Is an original way to say "I'm not connected to the phone". What I suggest is besides the vibration, show any message on the screen during 1 o 2 minutes (or even better, until reconnect or manually discarding the message)

closed 2021-10-02 by Avamander