InfiniTimeOrg/InfiniTime #598: Feature request - Find my phone option

AndriiZ, 2021-08-18:
It would be great to search lost phone from the watch. I dont know if any companion app support this feature

tmilburn, 2021-08-23:
Gadgetbridge has the required activity which can be triggered https://github.com/Freeyourgadget/Gadgetbridge/blob/df44b798c91d832d1ae05821c860decc5ba8ac93/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/activities/FindPhoneActivity.java but it requires code to be added to the pinetime device support in https://github.com/Freeyourgadget/Gadgetbridge/blob/d951ab1da7c4e08241eb25938f99cbe884fbb74d/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/pinetime/PineTimeJFSupport.java and a way of triggering this from the watch. I would suggest a new service for this with a characteristic to start and stop finding the phone.

Avamander, 2021-08-23:
> I would suggest a new service for this with a characteristic to start and stop finding the phone.

Standard has an immediate alert characteristic for exactly this purpose.

tmilburn, 2021-08-23:
Isn't the immediate alert service for finding a device (peripheral) rather than a phone (central) since it has a single write characteristic.

Avamander, 2021-08-23:
Both sides can expose services and characteristics.

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/543

closed 2021-10-02 by Avamander