InfiniTimeOrg/InfiniTime #599: Music app buttons don't work + swipe issue

hubmartin, 2021-08-18:
Hi, can anybody try music app with android please? I do not own other phone to test with.

In current develop ee44b6f the buttons don't work. When on debugger then it seems like the events LVGL aren't triggered at all.
When I tried tag 1.3.0 the buttons works reliably.

On top of that, in 1.3.0 and current develop ee44b6f when I do the swipe gesture for previous/next then it skips several songs. I can also see that selected song changes quickly few times. So it seems like it is sending multiple commands on single swipe, is this possible? (I do not have random song. When in 1.3.0 cklick previous/next button it skips correctly one song, when I do swipe then it skipps few of them, sometimes restarts watches).

Videos:
https://youtu.be/OYB3acYyOZ8 1.3.0 - buttons work properly, swipe skips multiple songs, sometimes freezes watches 
https://youtu.be/yw1J42DHpdY - develop branch ee44b6f, buttons do not work, swipe skips multiple songs, sometimes freezes watches

Riksu9000, 2021-08-18:
I found the cause of button presses getting ignored and fixed it in #600.

At least in Twos a single swipe has always been able to activate a couple of times. Recently it has gotten  worse somehow, but #492 should fix it.

closed 2021-08-18 by JF002

hubmartin, 2021-08-18:
Awesome! Thanks for quick fix and merge.

PR #600 (2021-08-18): Fix music app buttons
Music OnTouchEvent() also returned the wrong value, which caused the touch to get ignored after #426.

Fixes #599.