InfiniTimeOrg/InfiniTime #601: Clear Notifications

StuffOfSonny, 2021-08-19:
Ok so, after my first real day of serious use, I found myself wondering "how do I clear the notifications?" several times. The only hotfix I found was to reboot the watch, which if you ask me is kinda overkill. Is there already a built in way to clear them out, or is that on the TO-DO?

As per my personal philosophy, I don't like to complain about things without offering a solution (where I can help it), so my suggestion would be to long press a notification to bring up an option to remove the notification.

Riksu9000, 2021-08-19:
Duplicate of #176 and #272 and there is a PR that adds this feature #323

StuffOfSonny, 2021-08-19:
So does that mean this will be available in firmware 1.4? Because I had to flash 1.3 when I got the watch, and this does not work

Riksu9000, 2021-08-19:
> So does that mean this will be available in firmware 1.4?

Most likely.

adocampo, 2021-09-07:
Unfortunately, 1.4 release doesn't include this yet. Hopefully it will be present on the upcoming version :)

closed 2021-10-02 by Avamander