InfiniTimeOrg/InfiniTime #604: Quick navigate to most recently used app

marbetschar, 2021-08-20:
It would be convenient to have a way to quickly navigate to the most recent "active" app from the watch face. For example using a left-to-right swipe (from what I found, this is not used yet). A simple push on the button would then instantly go back to the watch face.

Personally I'd use this for the music app a lot - because I can't see the current time in the music app. But I can also see this to be useful for the heart rate monitoring and/or other potential future apps.

So maybe the "best" solution here is to provide one slot for the last used app which can be directly jumped into from the watch face using the left-ro-right swipe gesture.

hubmartin, 2021-08-21:
Not bad idea, I also use music app a lot. But #487 could solve that and you could configure it to any app. Maybe that PR could be later improved so you can choose the app or the "last app" option.

hubmartin, 2021-08-21:
Also #480 might have configurable long or double click in the future. That PR is working and currently there is no configuration, it is hardcoded now. This way we can run app by just button, no need to swipe.

coxtor, 2021-10-03:
#708 basically implements this , however is in a wip state

Riksu9000, 2023-02-20:
I'll close this in favor of fixing #1592.

closed 2023-02-20 by Riksu9000