InfiniTimeOrg/InfiniTime #607: Provision for manual time & date set

stefanoaz, 2021-08-21:
I'm presently running Infinitime V1.3 on PineTime. I have no way to synch the time from my iOS phone. Also, because my Linux machines run Debian (x86) and Raspberry Pi OS (ARM), the Linux companion apps are not available on these platforms. Although someone is working on an iOS companion app, my Mac OS version (10.15.7) will not support a version of the XCode IDE new enough to build his app source code to the target iOS.

It seems to me that it would be a good fallback option, to bake into the infinitime OS a simple UI to set the time and date manually. This would at least make the watch usable until more companion app solutions become available. Presently I have no way to set the time.

Riksu9000, 2021-08-21:
There is an open PR that adds this feature #472

Itai-Nelken, 2021-08-21:
Siglo works well on RPi

stefanoaz, 2021-08-21:
Where do I get Siglo for RPi?

Avamander, 2021-08-21:
> I have no way to synch the time from my iOS phone.

Raw characteristic writes work with nRF connect. Two companions are under development.

In general, time alone won't save you without a companion app.

stefanoaz, 2021-08-21:
I can use nRFConnect to upload firmware, but that's all. That's why the request for the fallback of manual set for time & date.

Itai-Nelken, 2021-08-21:
> Where do I get Siglo for RPi?

install it the regular way, its written in python so will run on any system with a python interpreter and the needed dependencies.
The flatpak also works.

stefanoaz, 2021-08-21:
So, when building the dependencies for Siglo, "gatt" does not appear to be available in the debian libraries using the apt package installer.

Avamander, 2021-08-21:
> but that's all

It is not.

Arsen6331School, 2021-08-21:
The daemon I made should work on Linux x86, arm32, arm64, and most other architectures. It can sync time and will do so when the PineTime connects.

https://gitea.arsenm.dev/Arsen6331/itd

Arsen6331, 2021-08-21:
The daemon I made can run on Linux on any architecture including x86 and ARM.

https://gitea.arsenm.dev/Arsen6331/itd

stefanoaz, 2021-08-21:
Arsen6331: for the "go" compiler, there is a package called golang that I can install - I assume that's the right one...

Arsen6331, 2021-08-21:
> Arsen6331: for the "go" compiler, there is a package called golang that I can install - I assume that's the right one...

Yes, that should be the right one as long as it's not too old.

Itai-Nelken, 2021-08-21:
> So, when building the dependencies for Siglo, "gatt" does not appear to be available in the debian libraries using the apt package installer.

Install python3-pip and then run `pip3 install gatt`

stefanoaz, 2021-08-21:
Arsen6331: I did the following:
make GOOS=linux GOARCH=arm && sudo make install

It built for a while, grabbing a whole bunch of things from google.com and google.golang.org. But it failed at the following step:

build go.arsenm.dev/itd: cannot find module for path io/fs

Arsen6331, 2021-08-21:
> build go.arsenm.dev/itd: cannot find module for path io/fs

That means your go compiler is too old. I could compile it for you, but I would understand if you didn't trust it. If so, you can use this guide for installing 1.16 which is new enough: https://tecadmin.net/install-go-on-debian/

stefanoaz, 2021-08-21:
@Itai-Nelken I got Siglo to install on the Pi. It cannot find the watch however. In my bluetooth settings, 
I can find the watch, but when I attempt to pair, the request times out. There doesnt seem to be a way on the watch side, to acknowledge a pairing request.

Arsen6331, 2021-08-21:
> There doesnt seem to be a way on the watch side, to acknowledge a pairing request.

Yes, you are right. I have found `siglo` to work best after you remove the watch from your bluetooth menu, then it can discover the watch.

stefanoaz, 2021-08-21:
I removed and rebooted - siglo still doesn't discover the watch. Also, the "Bluetooth settings" button on the siglo UI appears to do nothing.

Arsen6331, 2021-08-21:
> Also, the "Bluetooth settings" button on the siglo UI appears to do nothing.

It also does nothing for me currently. My frustration with siglo is partly what prompted me to create my daemon. I am currently making a CI pipeline for it which will build every release automatically for multiple architectures. I will add arm in there.

stefanoaz, 2021-08-21:
Siglo github page lists required libgtk-dev minimum version of 3.30, but version from debian repo is 3.24.

Arsen6331, 2021-08-21:
I have finished creating my CI pipeline. You can access the binaries here: https://minio.arsenm.dev/minio/itd/ and you can see the build logs here: https://ci.appveyor.com/project/moussaelianarsen/itd

stefanoaz, 2021-08-22:
@Arsen6331 How do I execute these binaries? Do both need to be running? In which order?

Arsen6331, 2021-08-22:
> How do I execute these binaries?

 Only `itd` needs to run for the time sync. `itctl` is for controlling `itd`, as seen in the readme. You can also get the systemd service file, itd.service (to be placed in `/usr/local/lib/systemd/user`), and the config file, `itd.toml`, (to be placed in `/etc`) from the repo. Then, run `systemctl --user start itd` **without root**, as seen in the "Starting" section of the readme.

stefanoaz, 2021-08-22:
I have a binary named itd-arm. I assume that I should re-name it to itd, and then place it in user/local/bin. Is that correct? Then I should invoke it using systemctl as outlined on your github page?

Arsen6331, 2021-08-22:
> I assume that I should re-name it to itd, and then place it in user/local/bin. Is that correct? Then I should invoke it using systemctl as outlined on your github page?

Yes, this is correct, but for systemd to start it, you need to place [`itd.service`](https://gitea.arsenm.dev/Arsen6331/itd/src/branch/master/itd.service) into `/usr/local/lib/systemd/user` and for the config options, you need to put [`itd.toml`](https://gitea.arsenm.dev/Arsen6331/itd/src/branch/master/itd.toml) into `/etc`.

stefanoaz, 2021-08-22:
Okay - the daemon is running and the files are in place. When I use the command 
"itctl get address", it gives me a socket dialing error and asks if itd is running (it is, I can see the process running in top).

My its.service file is in /usr/lib/systemd/user, with no /local/ in the path. That's where systemd was.

Arsen6331, 2021-08-22:
The socket is not running because it has not connected to InfiniTime. The watch only advertises for three minutes after the screen turns on if I remember correctly. Turn it on to start the advertising, and then run `systemctl --user restart itd`, which should send a notification to your PineTime saying it has successfully connected. If this does not work, try again after installing `playerctl`. That should not be required, but I have never tested it without `playerctl`, so that might be causing some weird issue.

stefanoaz, 2021-08-22:
Okay - the answer was to reset the watch - it waqsn't broadcasting its bluetooth even with screen re-awakenings. I had to do a hard reset of the watch, and it is now connected to itd. I can get the address, but getting the battery state gives "Error reading line from connection error=EOF."

BUT - it automagically set the time and date :) Great!

Arsen6331, 2021-08-22:
> getting the battery state gives "Error reading line from connection error=EOF."

Huh. Interesting. I only have one sealed PineTime to test on only one version of InfniTime, so I don't know how other watches may behave. I will need to investigate that further.

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/364

closed 2021-10-02 by Avamander