InfiniTimeOrg/InfiniTime #609: Rapid battery drain

joseph58tech, 2021-08-24:
I've noticed that ever since I upgraded to infiniTime 1.3, the battery has just drained, but it maybe because I hadn't charged the thing in 3 days
[image]

joseph58tech, 2021-08-25:
Alright, it was the low charge during upgrade

JF002, 2021-08-28:
Yes, this is a known behavior. It's caused by the non-linear discharge curve of the battery.
Battery level measurement will be improved in InfiniTime 1.4 :)

joseph58tech, 2021-09-02:
Alright thanks

JF002, 2021-09-02:
InfiniTime 1.4 is released, it should improve the battery level measurement. Feel free to provide some feedbacks when you try it!

joseph58tech, 2021-09-04:
I just download it, I will install in a little bit, but anyways thanks for understanding

joseph58tech, 2021-09-04:
Just installed infiniTime 1.4 and it is working flawlessly

pag-r, 2021-09-06:
Updated to 1.4, Battery level = 37 - everything fine. Day after - battery drained completely. Seems like to doesn't fully works

closed 2021-09-08 by joseph58tech