InfiniTimeOrg/InfiniTime #626: New vs already read notifications

drzacek, 2021-08-30:
It would be nice if new notification had different colors, background color, or frame, or some kind of indicator that it is new. 

Then combine this information with the new notification icon on the watchface - so it will display it until user read all new notifications.

This way we could drop the current behavior, where new notification is still there even after 30 minutes when watch is woke up. 

I think this is a bit annoying, because I want to be able to check time on my watch, and not always am able to dismiss the notification.

hamblingreen, 2021-09-01:
While this feature request has merit, the ability to dismiss notifications is in the 1.4 milestone ;)

JF002, 2021-09-02:
I decided to release 1.4 without the dismiss notification feature as the PR received no update for some time, but I hope the author will finish it soon :)
It might be possible to implement this feature request even if we add the dismiss notification feature, I think.

drzacek, 2021-09-02:
At least new notitications don't stick forever now.

kennysc, 2021-09-15:
Having the time when the notification happened would also be nice I think. Or someting like "30 min ago" maybe ?

Riksu9000, 2023-01-16:
Notification dismissal was added in 1.10.0. Is there still an issue or can this be closed?

closed 2023-02-21 by Riksu9000