InfiniTimeOrg/InfiniTime #627: Step counter "too generous"

marbetschar, 2021-08-30:
I found the step counter to be "too generous" with counting on my end. No clue what's causing this, but compared to the step counting algorithm in iPhones, the amount InfiniTime reports is about twice as high.

I'm running InfiniTime v1.3.0 on a Pine Time.

Avamander, 2021-08-30:
Tests have shown it to be very accurate, are you sure your iPhone is correct? Third reference here would be required to determine the device at fault.

marbetschar, 2021-08-30:
Nope, not sure. Its just a feeling of too much steps beeing counted by Pine Time. But its entirely possible I'm biased towards what I've got on the iPhone.

JF002, 2021-09-02:
The step counting is done in hardware , by the motion sensor. InfiniTime does not do the signal processing to estimate the number of steps, but it might be possible to tune the sensor to be more or less sensitive.
I can see 2 reasons why the step count *seems* more accurate on your phone : your phone probably have higher end sensor**s**, and you don't wear your phone on your wrist :). The sensor on the pinetime is always in motion  and probably interpret some of your arm motions as steps.

Personally, I consider this step counting feature more as a way to know how much I moved during the day than just the number of steps I walked.

[Here is an article](https://www.ncartron.org/pinetime-steps-counter-accuracy.html) showing that the step counting from the PineTime is quite good in comparison to 2 other smartwatches.

marbetschar, 2021-09-03:
> I can see 2 reasons why the step count _seems_ more accurate on your phone : your phone probably have higher end sensor**s**, and you don't wear your phone on your wrist :). The sensor on the pinetime is always in motion and probably interpret some of your arm motions as steps.

Yeah, I thought about those possibilities as well already.

> Personally, I consider this step counting feature more as a way to know how much I moved during the day than just the number of steps I walked.

That was my initial intention as well. I was just surprised to see that much of a difference between the two devices (like up to around +80% on the Pine Time).

> [Here is an article](https://www.ncartron.org/pinetime-steps-counter-accuracy.html) showing that the step counting from the PineTime is quite good in comparison to 2 other smartwatches.

Interesting read! I'll guess it all comes down to the fact we are counting steps on the wrist then.

closed 2021-09-03 by marbetschar