InfiniTimeOrg/InfiniTime #628: Security: Auth Request should be Pinetime-first

pdolinic, 2021-08-30:
Not everyone should be able to connect to the Pinetime to execute code. This is security risk, even though the watch has to execute updates. 

Solution: Enable Requests prior to code upload or connections.

JF002, 2021-09-02:
Indeed, there's not security feature in InfiniTime right now. NimBLE provides many security features, but we need to enable/intergrate them in the project.

evergreen22, 2021-09-07:
I've been working on encryption+authentication and bonding for a while. I have successfully enabled encryption and authentication with a pass key, but not bonding. Once I get bonding sorted out I'll make a PR.

Note that when the watch is connected, no one else can upload firmware because InfiniTime only supports one ble connection at a time.

I don't know what your threat model is, but there is not an easy way to prevent an evil maid attack. If you leave the watch unattended and unconnected anyone can get to it to modify the settings or firmware even if encryption+authentication is enabled for connections.

GIJack, 2021-09-11:
Threat models. Someone gets close with a handheld device and executes one of the following attacks

1. MITM  - can read messages, see who's calling, intercept bio-data like heartrate
2. Exploit against the phone - use address/call log permissions to attempt and read this data off the phone using gadgetbridge
3. exploit against the watch - impersonate the phone to upload and run code.
4. 4 de-auth and connect. jam/block/deauth the connection, or wait for one of the numerous disconnects the pinetime has and reconnect before your device has.

Most of these hinge on a handheld device that is a modified cellphone, or something of similar size. Devices like this are relatively cheap, and the physical methodology of such "bump" attacks with small concealed computers with wireless devices has been demonstrated against NFC devices.

If you as much make the complexity of the attack being you have to physically take possession of the device, and then scroll through the settings and remove the current pairing to add a new one, you've done a lot as its pretty feasible to keep the thing on your wrist most of the time. Also, if you pair it, and need to go into settings to remove the pairing, and can't add back an old pairing, it will be tamper evident.

Doing as much as bonding + encrypt + authentication that needs to be manually removed in settings before adding a new one, would go a long way. You'd increase the complexity needed greatly

pdolinic, 2021-09-12:
I entirely agree with @GIJack regardless if only 1 connection is allowed or not, pairing the watch with a phone is a giant security risk currently until the appropriate authentication is given, and this should be communicated publicly imo.

I personally would not grand that many rights over my phone as of now.

Avamander, 2021-09-12:
> I personally would not grand that many rights over my phone as of now.

Given the fact that all actions are implemented by the companion and it's the notifications that might have some impact, then disable those and be done.

closed 2021-12-09 by Avamander