InfiniTimeOrg/InfiniTime #630: Abstraction layer of lvgl widget

toitoinou, 2021-08-31:
I do not know a lot above lvgl but when I look at several application code I found that lots of code is duplicated and everything is very verbose.
Perhaps a more object oriented approach by creating a wrapper above lvgl could improve that.

Riksu9000, 2021-08-31:
In most cases duplicate code could be avoided by simply using loops, but they're not used in most apps for some reason. Similar buttons could be in an array and use an enum. This is what I did in #554. Are there other issues with how it is currently?

Avamander, 2021-08-31:
> but they're not used in most apps for some reason.

Because iterating over designs is very PITA with such code. The moment you want to change one single element amongst a bunch, you also run into annoyances.

Aside from using up a tiny bit of screen space, there's not a lot why such compaction would matter.

Riksu9000, 2021-08-31:
> Because iterating over designs is very PITA with such code. The moment you want to change one single element amongst a bunch, you also run into annoyances.
>
> Aside from using up a tiny bit of screen [...]

Most was probably the wrong word for me to use. Most objects are different and should be separate but there are some cases where I think using loops could be better. It can make the code easier to read as well.

Avamander, 2022-03-31:
Let's reopen this if we really feel that this has become a thing we really need.

closed 2022-03-31 by Avamander