InfiniTimeOrg/InfiniTime #632: Disable Bluetooth

convolutionbbs, 2021-09-01:
Hi - is there a way to disable bluetooth from the watch?  If not, could it be added as a feature request?

ajack2001my, 2021-09-01:
Until we can input date/time directly on the watch, I don't think it's a good idea.

JF002, 2021-09-02:
It's not possible to disable bluetooth for now, but it might be an interesting feature to add !

geekbozu, 2021-09-03:
We should definitely consider adding an "airplane" mode setting. Should not be to to difficult either.

davidhealey, 2021-09-03:
> Until we can input date/time directly on the watch, I don't think it's a good idea.

But you could use bluetooth to set date/time then turn off the bluetooth, no? It wouldn't be any different from not being near your phone.

GIJack, 2021-09-11:
> Until we can input date/time directly on the watch, I don't think it's a good idea.

It is. two reasons

1. airplane mode. So you can take this on an airplane, or wear it through areas that are RF sensative or can't have radios. It will still be usable as a watch, syncing before and after a flight.
2. storage. If you are not using the watch, and its not connected, you don't have to worry about it broadcasting anything or someone else connecting to it

Chasarr, 2021-09-12:
Restarting bluetooth is definitely a welcome feature now when it tends to be a bit unstable. I frequently have to restart my watch to reconnect it to my phone, and a simple "restart bluetooth" button would be great

Avamander, 2021-09-12:
@Chasarr 

> and a simple "restart bluetooth" button would be great

Nope, not a "simple" thing, safe to say it'd be here if it were.

Chasarr, 2021-09-12:
Oh, don't get me wrong! I meant clicking the button might be simple, not actually implementing it! Bluetooth connectivity seems to be working better with a pull request merged for 1.5 which might remove my need for such an option for me.

I was just explaining another use case of the button which wasn't mentioned here previously, so I would argue that I actually added *something* to the conversation, although not vital. I'm not very well versed in Github issue comment etiquette, so apologies if my comment felt like spam to you

tommytran732, 2021-09-13:
It would really be great if we can have secure pairing and an option to disable bluetooth :)

Avamander, 2021-09-13:
@tommytran732 

> It would really be great if we can have secure pairing and an option to disable bluetooth :)

Secure bonding is a separate issue and if you simply agree with a request, use the :+1: react.

medeyko, 2021-12-14:
There also exist potential tracking vulnerabilitirs when BLE is on, so maybe it's one more cause to have an option to turn BLE off:

https://www.theregister.com/2021/10/22/bluetooth_tracking_device/
https://cseweb.ucsd.edu/~nibhaska/papers/sp22_paper.pdf
https://www.zdnet.com/article/bluetooth-vulnerability-can-be-exploited-to-track-and-id-iphone-smartwatch-microsoft-tablet-users/
https://www.usenix.org/conference/usenixsecurity21/presentation/stute

closed 2022-02-20 by JF002

uis246, 2022-02-20:
Thanks. Non-advertising mode with keeping connections left, but maybe it is not so useful as complete disabling.

JF002, 2022-02-20:
The airplane mode feature should disconnect from any BLE device and stop advertising. In theory, it should prevent the radio from emitting any signal. I don't have the necessary equipment to check that in reality, though.

commit 319030d9 (2022-02-20): Add airplane mode feature

Implements 'Airplane mode' feature to disable and enable bluetooth/ble
Adds airplaneMode as a non-persisted setting
Adds a setting menu for switching airplane mode on and off
Displays an airplane symbol on the Digital watch face and the
    PineTimeStyle watch face when airplane mode is enabled
Always enables bluetooth/ble on boot (disable airplane mode)
Alphabetizes the settings menu options
Style cleanups

Closes #632
PR #888 (2022-02-20): Airplane mode feature (redesigned)
# Add airplane mode feature (redesigned)

- [X] Implement 'Airplane mode' to disable and enable bluetooth
- [X] Add a setting menu for switching airplane mode on and off
- [X] Add an airplane icon that is displayed when airplane mode is enabled

You turn on airplane mode by swiping and selecting settings, selecting *Airplane mode*, then choosing **Yes**. This will disconnect the watch if it is currently connected or cancel advertising if it is not connected. When airplane mode is on you cannot connect to the watch (i.e. bluetooth communications are disabled). [... cut]