InfiniTimeOrg/InfiniTime #640: Camera Controls

MaWalla, 2021-09-03:
When having the phone on a tripod or in some other fixed position (for example for group photos, or astrophotography), it would be very convenient to have a shutter button on the PineTime. A nice addition (if applicable) would be the possibility of having manual controls such as ISO, exposure time, maybe even the focus there too.
The icing on the cake would be the ability to see the camera viewfinder on the watch, similar to what is possible with an apple watch, even if only at like 1 frame per second. I assume that both BLE will be too slow and resources (CPU and especially RAM) won't be enough for this to be possible (next to the implementation on the phone being rather difficult I guess).

For this, a seperate camera app within InfiniTime could be used to provide the necessary interface, but also an expansion to the companion app, which shouldn't be too difficult for the shutter button I think, but probably for the rest, if not straight up impossible there, but others may be more aware of the possibilities here than me.

Long story short: it would be awesome to have a shutter button in InfiniTime, more camera controls/viewfinder would make this even better.

Avamander, 2021-09-03:
Streaming an image would in theory be doable. 

The bigger question is, are there any camera apps that are willing to receive external Intents for trigger and provide low-res viewfinder image trough a ContentProvider?

MaWalla, 2021-09-03:
This is great to know! (I have another unrelated idea I'll put into another issue)

Anyway, the question is indeed good. I don't know if it's related, but maybe you know these selfie sticks which used a 3.5mm audio jack cable with a button near the handle to trigger the shutter. Since those are a thing, there must be some kind of interface to achieve it. I don't know if this can be translated to the companion app using it though.

Avamander, 2021-09-03:
The audio jack ones tend to be TRRS, and emulate a headset button, it's usually not available to software.

Landrash, 2021-09-06:
Cheap Bluetooth controllers usually send **Volume Up** and I've used them with a bunch of applications on Android without a problem.

jmaris, 2022-04-04:
Sending volume up would likely be the best approach to this!