InfiniTimeOrg/InfiniTime #642: Ability to read full length notifications

calhix, 2021-09-04:
I know there is work on going for clearing notifications and further gesture support but it would be nice to be able to view the full messages recieved by maybe tapping once on the notification then scrolling down or something similar.

adocampo, 2021-09-23:
I don't know where is that discussion is being carried, or if it is internal to devs or open to the public, but I will give my two cents here.

 - I find the scrolling text annoying, most of the time is longer even than the full notification itself, instead of that text I would rather prefer the time and date of the notification and the subject and notification being in the text box area. Perhaps with two types or sizes of font (one for the subject and the other for the notification text itself), or separating the subject from the text with a horizontal line.
 - Switching notifications to next and previous could/should be swiping to right and left instead of up and down, so with up and down we could read a full-length notification. Perhaps notifications should be placed "on the right side" of the home screen, instead of "above", so swiping up and down would be consistent to read full length notifications.
 - Discarding current notification or all notifications will be of great use. Perhaps with double tap for discard current notification or when on the last notification, swiping right again a new screen would appear asking to discard all notifications (this swiping method would make sense if the former point would be operative)
 - When open notifications, it shouldn't open the first notification first, but the newest/unread one. 
Extra coolness
 - Sync read notifications to phone. Once I've read a notification, it should be discarded on the phone.
 - A visual way when browsing notifications to let the user know this notification has already been read, like changing the typo for a thinner font.
 - An icon of the notifying application would be great, but we can all live without it.

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/388

closed 2021-10-02 by Avamander