InfiniTimeOrg/InfiniTime #647: Adding display modes to PineTime

wilsonjholmes, 2021-09-06:
## Brief overview on how this came about: 
 
I initially submitted PR [#633](https://github.com/JF002/InfiniTime/pull/633) as I wanted to make the watch more astronomer-friendly by adding a way for the flashlight to be changed to red as a way to help preserve night vision when doing astrophotography and the like. That PR ended up raising the question of where individual app settings should be accessed on the watch: In a settings "pane", or through the app itself. That is kinda its [own discussion now](https://github.com/JF002/InfiniTime/pull/636#issuecomment-912052329), but as [tmilburn](https://github.com/tmilburn) and [geekbozu](https://github.com/geekbozu) brought up in the discussion on that PR, It might be better for the whole UI of the watch to be changed to a sort of "redscale" so that navigating the watch could also be done without messing up time acquired night vision/sensitivity.

## Where things are now

GeekBoy1011 on the discord helped me get the display buffer bit-mathed in DrawDisplay before it was actually drawn, and that was very promising, still very rough, but here is a proof of concept.
[image]
[image]
[image]
[image]
Currently only red colors showing, and any color that is not red is "converted" so you are still able to see things that are pure green or blue in the UI. This is in progress now and looking nice, but there are some issues thus far to think about.

## In need of ideas on the best way to move forward with this

This change has the possibility to wreck contrast in parts of the UI if certain colors are used in elements that are stacked on top of one another. A pure red will display the same as a "pure" purple. I am not sure what the best way to mitigate this is:
- While it may take more work I was wondering if it would just be better to change all of the UI to use defined color values, and those colors could be swapped out by a setting depending on what display mode you are in (e.g. Normal, Night-Mode, Greyscale)... Kinda like we already have `settingsController.GetPTSColor*()` in `PineTimeStyle.cpp`. We could, by this same method have a way for a completely custom set of colors to be defined for a redscale UI that would still have proper contrast...
- I have not really looked much into changing the gamma of the display, but I know that is another option to look into.
- We could just use the same bit masking/color conversion thing as show in the pictures and make sure that all colors in UI have significant enough contrast.
- I'm sure there are some things I have not though of yet... eager to hear input and maybe some other ideas.

Riksu9000, 2021-09-06:
I think the UI should have high contrast for usability and accessibility anyway, so just a filter should be good. Here's a color temperature conversion algorithm that can be used for "low blue light modes" https://tannerhelland.com/2012/09/18/convert-temperature-rgb-algorithm-code.html

evergreen22, 2021-09-06:
I like the idea of a 'night mode' in red that does not mess up night vision. I can think of other places this would be useful such as theatre, film, old school photography, and of course ninja.

The color settings are sprinkled all through the code now and it would be nice to have a common interface to set the screen colors. Perhaps a collection of themes that each 'screen' could index into to get the color setup. Then a color picker could just modify a 'user theme' for personalization.

wilsonjholmes, 2021-09-06:
> I like the idea of a 'night mode' in red that does not mess up night vision. I can think of other places this would be useful such as theatre, film, old school photography, and of course ninja.
>
> The [...]

I was thinking that would be slick, but I am new to the project, and would need some help understanding the different types of color categories to be used in a theme to cover all parts of the UI...

hubmartin, 2021-09-08:
With PWM backlight PR you can change backlight in range fron 0 - 10000, from darkness to full brightnness.
https://github.com/JF002/InfiniTime/pull/575