InfiniTimeOrg/InfiniTime #650: [BUG] Watch will vibrate throughout the phone conversation

ajack2001my, 2021-09-07:
When I get a phone call, the watch will vibrate. It will vibrate till the call ends. I can imagine the battery dying pretty quickly if I was on an hour long call... Annoying too...  Anyways, not sure if it's a bug or me not doing the settings right. Using InfiniTime 1.4.0 and GadgetBridge 0.59.2 on a Samsung Note 9.

calhix, 2021-09-07:
I believe a duplicate of #591.

MaWalla, 2021-09-07:
I'd like to add that I got a phone call today I couldn't take because I was driving the car. The watch kept the display on and vibrated as intended, but continued even after the call stopped until I pressed the button on the side. I guess that when the call is either taken or stopped, the watch should also stop vibrating.

I'm on GadgetBridge 0.59.2 too, but using a OnePlus 6T with crDroid (LineageOS based / Android 11)

evergreen22, 2021-09-07:
See the comment by @Riksu9000 on commit [1635bfa](https://github.com/JF002/InfiniTime/commit/1635bfafb3fe9dce1ab05102d1bfeac1eca81c66) in 1.4.0 for the close screen problem on a call.

ajack2001my, 2021-09-27:
Using InfiniTime 1.5.0 now, problem still persist... :/

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/591

closed 2021-10-02 by Avamander