InfiniTimeOrg/InfiniTime #651: [BUG] Inconsistant interface in InfiniTime 1.4.0

ajack2001my, 2021-09-07:
In most components of the watch, a simple slide down will bring the watch to the previous screen. However, the MUSIC and the TORCHLIGHT app does not behave that way.

ajack2001my, 2021-09-07:
The games PONG and 2048 as well as the timer (pacing) app does not conform to the slide down as well.

Riksu9000, 2021-09-07:
Most of these apps use the gesture for something else or the closing gesture is disabled so the app won't be accidentally closed. Flashlight could be fixed and Music could possibly be redesigned to not use vertical swipes.

hubmartin, 2021-09-08:
I also found no use to swipe in torch to "turn it off".  I would prefer to close it.

Itai-Nelken, 2021-09-08:
I think the torch should return to the quick settings menu when closed.

kennysc, 2021-09-10:
> I think the torch should return to the quick settings menu when closed.

I agree, it would seem more logical this way.

PR #720 (2021-12-30): Improve gesture consistency across the firmware
Metronome can be closed by swiping down from the center. I could make it not close while dragging the arc, but I couldn't make it not close while dragging the dropdown menu, so this is the next best thing.

Music app shows volume buttons on swipe up and swiping down goes back to next/prev. now swiping down in the default state will close the app.

Flashlight can now be closed by swiping down. Also the transitions animation was changed to match other apps and returnapp is quicksettings instead of watchface.

The last apps that can't be closed thro [... cut]
closed 2021-12-30 by JF002