InfiniTimeOrg/InfiniTime #653: [Feature Request] Fuzzy clock watchface

adocampo, 2021-09-07:
Coming directly from Pebble, which died last month suddenly, I was very used to a fuzzy clock watchface.
I would love to see it on the PineTime as well. Just a plain, fuzzy watchface. No date, no stats, no nothing at all, just the fuzzy time. Also, it would be great to have it in more languages, not just english.
What do you think?

adocampo, 2021-09-08:
Yes, surely it would better not to bloat the firmware with watchfaces, but as I'm pretty new to PT, and after upgrading the firmware the first thing I noticed was it has a new watchface, besides it has 2 games and other stuff I wouldn't say it's relevant for a firmware, I expected this feature request would fit here, at least until there is no way to "install" watchapps.

ajack2001my, 2021-09-08:
True. Can the devs let us know if this feature is already in the pipeline? :P

hubmartin, 2021-09-08:
Since I had to google it, I'll spare time of others. This is it?
[image]

geekbozu, 2021-09-08:
This watch face it self may be under work by someone. I have no idea. We do know we want better handling of watch faces and such. Slowly we are building up things to support this. Work for better watch face settings and other features are being added. So all in good time!

dyamon, 2021-09-09:
I saw this feature request yesterday and decided to give it a go since it's simple enough and will help me understand a bit better how the codebase works when it comes to apps and such. I'll keep you posted!

I guess the code can be moved away from the firmware itself once we have a solution for that.

For internationalisation, I'd wait for #417 and see what comes out of it.

adocampo, 2021-09-10:
> Since I had to google it, I'll spare time of others. This is it?
> [image]

Yes, that's the brilliant fuzzy clock watchface.

> I saw this feature request yesterday and decided to give it a go since it's simple enough and will help me understand a bit better how the codebase works when it comes to apps and such. I'll keep [...]

It's really great someone can catch this feature. I think it was one of the most loved features on the pebble. I really appreciate your time and effort and will follow up the progress.

Yes, I noticed all dead keys (like accents) are simply missing, and I figured out i18n wasn't here yet. Only bear in mind when developing the watchface to be multilingual in the future ;)

Great news, indeed :)