InfiniTimeOrg/InfiniTime #654: Now on 1.4 the auto connect problem is worse, now I have to delete the device from gadget Bridge and and reconnect it when my phone restarts, EITHER THE GADGET BRIDGE DEVOLPERS OR INFINITIME DEVOLPER NEEDS TO FIX THIS, it is a crucial problem

joseph58tech, 2021-09-09:


ajack2001my, 2021-09-09:
@joseph58tech Are you using Gadgetbridge 0.59.2?

joseph58tech, 2021-09-09:
Yes

chloecormier, 2021-09-09:
I've had the opposite experience, somehow. With 1.3, I had to remove the device from Gadgetbridge and re-pair it after some time had passed. Now with 1.4, it seems to be reliably staying connected.

In either case, I don't think 1.4 even touched the Bluetooth code, did it? Seeing any change between the two versions is admittedly very weird.

toitoinou, 2021-09-09:
@joseph58tech I have to do the same since I have the pinetime. So same thing in 1.3 and 1.4 for me.
There are currently working on these problems.

Avamander, 2021-09-09:
Duplicate of https://github.com/JF002/InfiniTime/issues/302

InFerYes, 2021-09-18:
> Duplicate of #302
> 
> Search before you open a new issue.

That issue is about switching to airplane mode and then being unable to connect. I think this issue is about losing connection seemingly "randomly". I have this issue. Watch or phone will just stop syncing all of a sudden while the BT adapter was never switched off. It just stops. When I restart the Watch, sometimes it still won't connect and I have to do an annoying dance of re-pairing the watch and restarting it.

Avamander, 2021-09-18:
@InFerYes It's the same issue.

InFerYes, 2021-09-18:
@Avamander You mean it's the same underlying problem.

joseph58tech, 2021-09-29:
OK so infiniTime 1.6 is now released claiming that the BLE was fixed so I will test that out and than close this problem

Bugsbane, 2021-09-30:
Sadly, I still seem to be getting this issue with Infinitime 1.6 and Gadgetbridge 0.59.3  ಠ~ಠ

Bugsbane, 2021-09-30:
Why did this get closed when there's still cases of it happening?

Avamander, 2021-09-30:
> Why did this get closed when there's still cases of it happening?

Because he doesn't have the issue any more.

You can create a new issue that describes exactly on which firmware version, what are the symptoms.

joseph58tech, 2021-10-01:
Sorry for the inconvenience I just experienced half of the original problem and reopened the problem

Avamander, 2021-10-01:
Now it's certainly a duplicate of https://github.com/JF002/InfiniTime/issues/302

joseph58tech, 2021-10-02:
> Now it's certainly a duplicate of https://github.com/JF002/InfiniTime/issues/302

I can see you've been waiting to say thay

closed 2021-10-02 by Avamander