InfiniTimeOrg/InfiniTime #656: [BUG} PineTime Style watchface step counter not working...

ajack2001my, 2021-09-10:
The PineTime Style watchface step counter (the circle on bottom right of screen) is not working. Can somebody else confirm?

dyamon, 2021-09-10:
How is it not working exactly? To me it wasn't even obvious that was the step counter indicator 😆

At the moment I have ~600 steps and the indicator is pointing at 1 o'clock.

ajack2001my, 2021-09-10:
What indicator? :)

[image]

Riksu9000, 2021-09-10:
It's not visible because the indicator is the same color as the background.

ajack2001my, 2021-09-10:
@Riksu9000 yes, you are right... apologies for the confusion... I feel a little stupid now.... :p

Riksu9000, 2021-09-10:
Well I think it should still be considered a bug that should be fixed so this issue is fine.

kieranc, 2021-09-11:
I'm inclined to agree, I'd like to address it in my next update to PTS to make the gauge needle use the existing colors array and also change color in case the background is the same as the needle.

PR #718 (2021-12-30): Integrate color picker into PineTimeStyle watchface
This PR aims to move the color picker functionality into the watchface, to avoid cluttering the settings list and potentially add color settings to other watchfaces.
The settings are now accessed with a long tap on the watchface, which shows a button - if the button is pressed within the timeout (currently 5 seconds), the color picker interface is loaded and can be used to change the colors as before. Otherwise the button disappears, this is to hopefully avoid accidental activation.
I plan to add a bit of logic to change the step gauge needle [... cut]
closed 2021-12-30 by JF002