InfiniTimeOrg/InfiniTime #659: Feature Request: Adding a notification when battery is low

franzcor, 2021-09-10:
It would be nice to receive a notification with vibration to alert you that the PT could shutdown soon due to low battery.
This could happen around 10-15% of battery level. Sometimes I just forget that the PT is running for days and it happened that I was wearing it and it was already fully discharged. 

The UI is not that clear about having low battery percentage even if it shows you that the battery is discharging, it's not easy to see if the battery is at 40 or 30 or 15% only from the icon. You are forced to check in the battery menu.

adocampo, 2021-09-23:
One of the coolest features of the Pebble, and that was working perfectly, was it warned the user an estimated timeframe before powering down.

[image]

It gave peace of mind to the users letting them know they still had like 12h or the like to charge the watch again. Way more cool and "human" than having a percentage.

Riksu9000, 2023-03-09:
The battery percentage reporting is improved in the next release. Also low battery will be indicated by making the icon orange.

closed 2023-03-09 by Riksu9000