InfiniTimeOrg/InfiniTime #661: feature request: watch to watch sync

GIJack, 2021-09-11:
Here is an idea: watch to watch sync to make sure everyone in a group has the same time. Watch 2 Watch sync.

Put an option in settings to broadcast a timesync command. Add another option that opens a dialog and waits for said broadcast, sets time, and goes back to the watchface.

Riksu9000, 2021-09-12:
Most devices connected to the internet get the time from a time server, thus everyone should have the same time, and that time is synced to the watch via a companion app. I haven't noticed the time ever drifting.

Have you encountered a situation where this would be useful?

Riksu9000, 2022-01-22:
You can now also set the time manually on the watch. I don't think there's much demand for this feature.

closed 2022-01-22 by Riksu9000