InfiniTimeOrg/InfiniTime #663: [Feature Request] Back button for 2s

calhix, 2021-09-11:
I'm a huge 2048 fan and use the back button a lot when playing on my phone/tablet! This would be a nice addition I think, though perhaps a bit subjective.

schneidr, 2021-09-12:
Could you describe that a little more? I have absolutely no idea what you mean.

calhix, 2021-09-12:
> Could you describe that a little more? I have absolutely no idea what you mean.

Sure, at least with the app I use (https://play.google.com/store/apps/details?id=com.tpcstld.twozerogame&hl=en_US&gl=US), this has a back button on the top right of the screen that's to the left restart button, you can see it in the first picture to the right of the video. It's nice when accidentally swiping the wrong way or if two swipes are registered when you only meant to do it once

schneidr, 2021-09-13:
Ah. I was confused because you call the game once "2s" and once "2048". And I was wondering how a back button for two seconds cold be helpful.
You should stick to one name. And maybe call it "undo" instead of "back", that would be crystal clear.

calhix, 2021-09-13:
Ah, the reason for the name change is I thought the game is named "2s" on the watch and it's called 2048 everywhere else, I said 2048 to clarify in the post I was talking about the game on other devices, didn't even cross my mind for some reason that 2s could be considered 2 seconds.

adocampo, 2021-09-13:
Not very related to this issue, but yesterday I got a game completed and, well, I expected something like "you won!" or the like, but in fact, after making the photo, I was be able to continue gaming... or the rules I know for 2048 are others or the objective of the game is to sum 2048.
[image]

Avamander, 2021-09-13:
@adocampo Most 2048 games allow continuing past the specified number, some also provide extra tiles at some point. It's pretty a feature and not the topic of this issue really.

calhix, 2021-10-17:
> @adocampo Most 2048 games allow continuing past the specified `2048` tile, some also provide extra tiles at some point. You just reached the score, not the tile. It's very much a feature and not the topic of this issue really.

This is correct, though I do agree a You Won! type of message would be nice to see upon reaching 2048 as I see in a lot of the 2048 apps and then you can just continue the game by swiping in any direction or just tapping the screen and then continuing.

Riksu9000, 2023-02-20:
There's not really space to put an undo button on the screen. How would the undo feature be accessed?

calhix, 2023-03-05:
> There's not really space to put an undo button on the screen. How would the undo feature be accessed?

I've been thinking that this could be done as a double-tap gesture, what do you think?