InfiniTimeOrg/InfiniTime #665: Wake on wrist working just on standing or sitting position

adocampo, 2021-09-12:
I noticed when I'm on the bed or playing with my baby, when I'm leaning over him, wake on wrist doesn't work. For wake on wrist to work, the watch plate must be facing the ceiling/sky and not any other direction. So it just works when you're in a standing or sitting position.

It would be nice if it can wake the watch' screen in any starting position and then tiling 90º in any Y direction

InFerYes, 2021-09-18:
It would probably trigger too much in that case I think.

adocampo, 2021-09-18:
> It would probably trigger too much in that case I think.

I don't know if PT's accelerometer can respond to "shakes" and not just "inclinations", so with it would trigger less accidentally.

Another idea if the former is not possible with PT's accelerometer, is set another wake up screen option which let the user choose between the current "raise wrist" and "tilt wrist", so people like me who prefer to see the time _in any position_ (for example, laying in bed) surely can live with accidental screen wake-ups.

darkdragon-001, 2021-10-08:
Related #691