InfiniTimeOrg/InfiniTime #666: Customised PineTimeStyle watchface, display glitches

alwayslurking, 2021-09-13:
I upgraded to 1.4.0 and decided to use the new watchface with some non-default colours. After that change, I suffered intermittent display glitching, blanking and partial redrawing multiple times a day. It feels like the timings may line up with notifications from my phone, but that might be me seeing patterns in coincidence. 

I've now moved back to the digital watchface and the misbehaviour hasn't reproduced in nearly two days, suggesting pinetimestyle is involved.

kennysc, 2021-09-13:
I'm not sure if it's related, but mine also gets glitchy sometimes. There are swipes and taps registered when I'm not touching the screen.
But it's also happening with the digital watchface, so the watchfaces dont seem to impact the behavior. Maybe it's a bug in the touch driver ?
I am also on 1.4.0.

Riksu9000, 2021-09-14:
Any more information on when exactly the display glitching happens? Does it happen on the watch face or some other screen? Could it be related to #445?

kennysc, 2021-09-14:
In my case there is no corruption of the image, so I doubt it's related to #445. 
there is no exact moment when it happens, sometime the watch is closed end it will open on it's own and change menu and open a random application. Other times I will be in an app and taps or swipes will be registered without me touching the screen.

I'm thinking that it's either the touch driver or the watch it self thar might have a problem.. should i communicate directly with pine64 ?

Riksu9000, 2021-09-14:
@kennysc 
> In my case there is no corruption of the image, so I doubt it's related to #445.
> there is no exact moment when it happens, sometime the watch is closed end it will open on it's own and change menu [...]

Your issue seems unrelated to this issue, as the original issue has to do with the display, and yours has to do with touch.

If you have raise wrist wakeup mode enabled, then it can sometimes unintentionally wake up and register some accidental touches while you're not looking at it. #548

If it's registering random touches while you're just looking at it and not touching it, then I think you should open a new issue.

But if this also happens before version 1.4.0, then I would think it's an issue with your watch.

alwayslurking, 2021-09-14:
After I raised the ticket, I had similar issues with the digital watchface, though it's been generally behaving since I took this:

https://photos.app.goo.gl/FyP9Y28xpEvGG6QX7

Riksu9000, 2022-02-11:
@alwayslurking Is this still an issue? It doesn't seem like others are having this exact issue, so it might be unique to your device.

Riksu9000, 2022-03-23:
I'll have to assume this is caused by a faulty watch. Let's reopen if new information comes up.

closed 2022-03-23 by Riksu9000