InfiniTimeOrg/InfiniTime #671: [Feature request] Timer optimization

Pepperspraylaugh, 2021-09-15:
I think a good way of optimizing the timer app is to allow the user to set HH:MM:SS of the timer via the same circlet meter the metronome app uses, with them being able to swap between setting each magnitude with a swipe. Simply moving the finger is far more efficient then just repeatidly pressing a button or having to hold it down and then reset it if you mess up.

Riksu9000, 2021-09-15:
A couple of other ideas I had for this are:
- Dragging the number up and down with a vibration for each click. LVGL Roller can't be used because it is too choppy.
- Having a "trackpad" where we can slide left and right to increase and decrease the time. With this we could see the number without our finger blocking the way.

I'm personally not a big fan of the arc in metronome because it is hard to set it to an exact value and the arc shape is kind of weird to drag and follow.

medeyko, 2021-12-16:
I agree that dragging the numbers up and down would be cool.

On the other hand, I'm quite unsure that more drastic changes in the interface will make it better. Like @Riksu9000 , I'm not a big fun of the metronome arc too...

Riksu9000, 2023-02-21:
Preset timers can be used to solve this issue, as suggested in https://github.com/InfiniTimeOrg/InfiniTime/issues/1409#issue-1430639110.