InfiniTimeOrg/InfiniTime #672: [Discussion] Battery should be an app

minacode, 2021-09-15:
The battery state is found in the settings, but for me, it feels a lot like Steps (at least, both are just a percentage meter). 
It also is no real "setting", because there is nothing to set. 

I do not mean this disrespectful. I just stumbled across this a number of times and it still feels odd to me. 
Maybe this can be a little discussion :)

escoand, 2021-09-15:
@minacode Nice idea. And this app could show some more information, like "time of disconnect from power" and "time left".

minacode, 2021-09-15:
Nice idea. I guess #585 would be helpful for that.

schneidr, 2021-09-18:
I disagree. Battery state is found in the settings in about every OS, that's where most people will look for it. Granted, more details would be nice, but it should stay where it is.

closed 2022-01-30 by minacode