InfiniTimeOrg/InfiniTime #675: [Feature request] Bluetooth switch under settings

adocampo, 2021-09-18:
As per issue https://github.com/JF002/InfiniTime/issues/302 and https://github.com/JF002/InfiniTime/issues/654, my PT disconnects sometimes from GB.
It can happen when I'm going away of my phone or even sitting beside it, I couldn't really reproduce the error as sometimes I can be an entire day connected and suddenly being disconnected and not get reconnected even if I'm beside my phone.

The only way to make it work again is turning off PT and turning it on back.

That made me think that perhaps, it could be of use to have a toggle Bluetooth settings in order to avoid powering off PT, as for now, if we reboot the device, we lose all settings, alarms, notifications, and others.

It would be useful also to let unpair the device from the watch settings, but that would be an extra feature, the important here would be let PT turn off/on Bluetooth, simulating the reboot of the antenna.

Avamander, 2021-09-18:
Duplicate of https://github.com/JF002/InfiniTime/issues/632

adocampo, 2021-09-19:
Damn it! I didn't see it, sorry.

closed 2021-09-19 by adocampo