InfiniTimeOrg/InfiniTime #683: Cannot download the 1.4 .zip file (XML error?)

bhibb, 2021-09-20:
When trying to download the latest release of the Infinitime 1.4, when I select the link for the .zip file, I receive the following error from Github. Does it need to be re-uploaded?
![image](https://user-images.githubusercontent.com/33728168/134077520-da489ce1-bbfb-4342-9d30-9b38d1c83c31.png
[image]
)

Avamander, 2021-09-20:
Probably a GitHub bug, try again later.

bhibb, 2021-09-20:
I hope so. It seems to be a broken link and companion apps like Siglo which fetch it are also broken. :*(

bradman117, 2021-09-20:
Thats odd. I am experiencing same issue. v1.3 download still works.

bhibb, 2021-09-20:
In the meantime until it's fixed, maybe someone who's already downloaded it can host it?

bhibb, 2021-09-21:
The error appears to be the following:

```
SignatureDoesNotMatch
The request signature we calculated does not match the signature you provided. Check your key and signing method.
```
Looks like maybe something was off with the uploaded signature.

BCNelson, 2021-09-21:
It is working for me now.

bhibb, 2021-09-21:
Confirmed. It's working now! I'm closing this issue!

closed 2021-09-21 by bhibb