InfiniTimeOrg/InfiniTime #684: Generalise "PTS Colors" into "UI Colors" menu

dyamon, 2021-09-22:
# Proposal

It would be better to turn the "PTS Colors" menu into a more generic menu to choose colors for InfiniTime.

# Description

The menu should be turned into, let's say, 4 items to set:
- background color
- background accent color
- foreground color
- foreground accent color

We can then make existing apps and UI dependent on these settings.

# Advantages

- future UI/app developers don't need to worry about coming up with custom palettes
- better UI consistency
- higher degree of customisation for the end user

schneidr, 2021-09-22:
Possible duplicate of #584

Riksu9000, 2021-09-22:
The accent color idea in #584 is to give the user a simple choice with all safe options. Having so many options probably won't work in practice.

What would background, background accent, foreground and foreground accent colors actually be?

>future UI/app developers don't need to worry about coming up with custom palettes

Developers would have to make sure all screens are usable with all color combinations, or just not be creative.

> better UI consistency

If the entire UI looked the same, I guess it could be called consistency, but it would also be boring.

> higher degree of customisation for the end user

This comes at the cost of making customization and designing overly difficult for everyone.

Also the black background is important for the UI as it blends the background and the screen border.

dyamon, 2021-09-22:
> What would background, background accent, foreground and foreground accent colors actually be?

I was thinking something like background = standard background color, background accent = alternative bg color that needs to "pop out", foreground = standard foreground color, foreground accent = highlighted foreground.

I'm not pushing for these 4 option but they seemed reasonable to me and easy to integrate in the guidelines.

> Developers would have to make sure all screens are usable with all color combinations, or just not be creative.

If we take phone apps, for example, I've never seen anyone trying to guarantee this. I can easily set white text on white bg on most of the "customisable" apps I have installed. This is also the case for PTS, btw.

This is also a common approach used by any software that supports "colorschemes". The app defines _color classes_ (color semantics), and the developer **should use the classes according to the provided semantics**, ignoring the actual colors that the user might use.

If the user wants to customise colors they will take care of inconsistences, taking into account _personal_ usability and color preference.

Also, creativity is not throwing color at the screen. The developer is not *forced* to use these colors ofc but they are guidelines (also to include [here](https://github.com/JF002/InfiniTime/blob/develop/doc/ui_guidelines.md)) and to make new apps "blend" better with the OS.

> If the entire UI looked the same, I guess it could be called consistency, but it would also be boring.

I'm not coming up with some crazy idea here. It is well know that consistency is a solid principle of good UI/UX. Even without considering color customisation, the "final product" will need to converge to a well defined palette of colors and other UI design principles (one example is transition animations, which are all over the place atm).

> This comes at the cost of making customization and designing overly difficult for everyone.

I disagree, customisation is there for the end user (if they want to dive into that, otherwise sound defaults should be enough), designing is simplified since you have additional guidelines about colors (along with the others)

> Also the black background is important for the UI as it blends the background and the screen border.

This is a personal preference, unless you want to make it into a UI guideline.

Riksu9000, 2021-09-22:
>If we take phone apps, for example, I've never seen anyone trying to guarantee this. I can easily set white text on white bg on most of the "customisable" apps I have installed. This is also the case for PTS, btw.

That doesn't make it okay to allow broken configurations IMO.

>This is also a common approach used by any software that supports "colorschemes". The app defines color classes (color semantics), and the developer should use the classes according to the provided semantics, ignoring the actual colors that the user might use.

What you were originally suggesting wasn't a theme or a colorscheme, but basically a theme creator on the watch. If we have a theme creator in the firmware, we should make sure it always produces good results, especially if we care about UX. It would be okay if the themes are created outside of InfiniTime, because then we wouldn't be responsible for bad configurations.

> I'm not coming up with some crazy idea here. It is well know that consistency is a solid principle of good UI/UX. Even without considering color customisation, the "final product" will need to [...]

I'm obviously not arguing that the UI should be inconsistent, but having everything configurable separately will lead to bad configurations and therefore inconsistencies. Still having custom themes is fine, because then it isn't our responsibility to make sure they are consistent.

>>Also the black background is important for the UI as it blends the background and the screen border.
>
>This is a personal preference, unless you want to make it into a UI guideline.

The black background makes it possible to use the entire screen space for objects without it looking too cramped. I suppose it could be added as a guideline if it matters.

The accent color selector would be simpler for both developers and users and there would be no wrong configurations, which will make it much more approachable than a theme creator.

dyamon, 2021-09-22:
I agree that the theme creator should not be part of the firmware necessarily. It should be an app and it seem the general consensus that eventually apps will be shipped outside of the firmware (?). But atm there is already a theme creator in the firmware and it works only for PTS. My suggestion was to make it more generic.

Still, if this is a "wontfix" type of situation or people think it's better to wait for this, we can close the issue.

Riksu9000, 2022-01-21:
The color customization has been integrated to the PTS watchface, so this is no longer relevant.

closed 2022-01-21 by Riksu9000