InfiniTimeOrg/InfiniTime #686: Bluetooth disconnecting everyday

SteeveGL, 2021-09-23:
Since I received my watch, every day the Bluetooth disconnect. I do a reset (restart) then the connection come back.

V 1.4.0
Using Gadgetbridge on Android.

arvchristos, 2021-09-23:
This happens to me too. I am using v1.4.0 and Gadgetbridge as the OP.

LlamasJM, 2021-09-23:
Me too. Using v1.4.0 and Gadgetbridge. Not necessary reset the watch. Only quit PineTime from Gadgetbridge, wait some minutes, and search and connect again.

bhibb, 2021-09-23:
This happens to me, too, but with the Infini-iOS app on iPhone. I wish there was a way to set the watch into a Bluetooth searching mode to have it start looking for devices on demand.

imp667, 2021-09-24:
Same issue here too. The update is really huge improvement but like others every day I have to remove the pinetime the readd and the. Disconnect then connect and things work.

Paggios, 2021-09-24:
yes, same problem. This issue appens also using wasp os?

serjor, 2021-09-24:
Hopefully, I'm not alone with this problem :disappointed:

Avamander, 2021-09-24:
Duplicate of https://github.com/JF002/InfiniTime/issues/302

@serjor If you just have the same issue, use reacts instead of comments.

SteeveGL, 2021-09-24:
> Duplicate of #302
> 
> @serjor If you just have the same issue, use reacts instead of comments.

It's not related

Avamander, 2021-09-24:
@SteeveGL it is the same issue.

Avamander, 2021-09-24:
The very least it's a duplicate of https://github.com/JF002/InfiniTime/issues/654, but that's again still the same underlying issue.

SteeveGL, 2021-09-24:
> @SteeveGL it is the same issue.

No, it has nothing todo with the airline mode. 

Before 7am works find. After 7am Bluetooth disconnect. Required reset.

Avamander, 2021-09-24:
@SteeveGL Try to ignore the title for a second, it hasn't been updated and it's user-chosen. In any case, use search before opening a new issue.

SteeveGL, 2021-09-24:
> @SteeveGL Try to ignore the title for a second, it hasn't been updated and it's user-chosen. In any case, use search before opening a new issue.

I did. Not duplicate

joseph58tech, 2021-09-27:
This problem is still ongoing in v1.5, just so you know

Avamander, 2021-09-27:
As the changelog doesn't say, the update doesn't contain any fixes to this issue.

SteeveGL, 2021-09-27:
I notice 2 distincts issues.

1. The watch disconnect by itself. Happening every day. Reset fix the issue. 
2. Gadgetbridge disconnect the watch. Happening randomly. Reset the watch doesn't fix the issue. Quitting Gadgetbridge doesn't fix the issue. Removing the watch from Gadgetbridge and adding it back fox the issue.

I'm talking about #1 here.

SteeveGL, 2021-09-28:
> @SteeveGL Does this issue always occur at the same time of day e.g. 7am? On how many consecutive days has this occured?

No it's not fix on a specific time. But must of the time it happens in the morning. It's close to be every 24h after the last reset.

> Issue #302 which will likely be fixed by #688 means that if the watch disconnects at any point after roughly 18.2 hours (e.g. moved out of range of the phone or phone drops the connection) then the watch no longer sends out bluetooth advertisement packets and has to be reset before being able to connect over bluetooth again.

Since I have my watch, I never been out of range very long.

tmilburn, 2021-09-28:
If you have the [nRF connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_GB&gl=US) then you can debug the bluetooth connection and work out the reason for the disconnection. This may be for many reasons not just out of range.

SteeveGL, 2021-09-28:
> If you have the [nRF connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_GB&gl=US) then you can debug the bluetooth connection and work out the reason for the disconnection. This may be for many reasons not just out of range.

```
nRF Connect, 2021-09-28
InfiniTime (F8:8A:34:AE:41:43)
V	09:01:07.440	Connecting to F8:8A:34:AE:41:43...
D	09:01:07.440	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	09:01:07.474	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	09:01:07.474	Connected to F8:8A:34:AE:41:43
D	09:01:07.474	wait(1600ms)
V	09:01:09.074	Discovering services...
D	09:01:09.074	gatt.discoverServices()
D	09:01:09.105	[Callback] Services discovered with status: 0
I	09:01:09.105	Services discovered
V	09:01:09.133	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Firmware Revision String [R] (0x2A26)
- Hardware Revision String [R] (0x2A27)
- Software Revision String [R] (0x2A28)
Current Time Service (0x1805)
- Current Time [R W] (0x2A2B)
Unknown Service (00000000-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [N] (00000001-78fc-48fe-8e23-433b3a1942d0)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (00000002-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000004-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000003-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000005-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000006-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000007-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000007-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000008-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000009-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (0000000a-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (0000000b-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (0000000c-78fc-48fe-8e23-433b3a1942d0)
Unknown Service (00010000-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010001-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010002-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010003-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010004-78fc-48fe-8e23-433b3a1942d0)
Alert Notification Service (0x1811)
- New Alert [W] (0x2A46)
- Unknown Characteristic [N] (00020001-78fc-48fe-8e23-433b3a1942d0)
   Client Characteristic Configuration (0x2902)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Immediate Alert (0x1802)
- Alert Level [WNR] (0x2A06)
Heart Rate (0x180D)
- Heart Rate Measurement [N R] (0x2A37)
   Client Characteristic Configuration (0x2902)
D	09:01:09.133	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	09:01:09.136	gatt.setCharacteristicNotification(00000001-78fc-48fe-8e23-433b3a1942d0, true)
D	09:01:09.137	gatt.setCharacteristicNotification(00020001-78fc-48fe-8e23-433b3a1942d0, true)
D	09:01:09.137	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
[... 177 lines cut ...]
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Immediate Alert (0x1802)
- Alert Level [WNR] (0x2A06)
Heart Rate (0x180D)
- Heart Rate Measurement [N R] (0x2A37)
   Client Characteristic Configuration (0x2902)
D	09:25:10.433	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	09:25:10.435	gatt.setCharacteristicNotification(00000001-78fc-48fe-8e23-433b3a1942d0, true)
D	09:25:10.436	gatt.setCharacteristicNotification(00020001-78fc-48fe-8e23-433b3a1942d0, true)
D	09:25:10.437	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
D	09:25:10.437	gatt.setCharacteristicNotification(00002a37-0000-1000-8000-00805f9b34fb, true)

```

Disconnected then I did a reset.
> D	09:23:43.559	[Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E	09:23:43.560	Error 8 (0x8): GATT CONN TIMEOUT
I	09:23:43.560	Disconnected
D	09:23:43.597	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	09:25:08.542	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	09:25:08.542	gatt.close()

Close to be exactly 24h since my last reset. I will let the debug running to see tomorrow to have the exact timestamp.

JF002, 2021-09-30:
InfiniTime 1.5 and 1.6 brought many improvements the BLE connectivity. While I doubt those changes fix **all** connectivity issues, they certainly fix most of them. I think this one (BLE that disconnects after a day) is one of them. I'll close this issue, but feel free to open a new one if you encounter another BLE issue on InfiniTime 1.6 (and if it doesn't exist yet).

closed 2021-09-30 by JF002