InfiniTimeOrg/InfiniTime #687: Add ability to set time if not connected to phone

michaelharmonart, 2021-09-23:
Feel like it should be a bare minimum that you be able to set the time on a watch, without requiring connecting to a Bluetooth device.

Would love to have the ability to set the time and date without connecting to another device.

bhibb, 2021-09-23:
Agreed! I saw this feature on Wasp-OS and thought it was a really neat idea. I do find it a little annoying that when my watch crashes or runs out of battery, that I have to pair it with a companion app to set the time.

adocampo, 2021-09-23:
I miss this feature too. It was on pebble and I founded it useful, but I wasn't aware of how much useful it is until I had my PT on my wrist, ran out of battery or get unpaired.

I think a smartwatch should be a standalone watch in first place, and have all the essential features a watch must have: time, date and alarms are the bare minimal, and they should be completely offline, or at least, have the option to be completely offline, in my opinion. 

Yes, if they can be set on the companion app is _also_ great, but one shouldn't be forced to be near a smartphone, or make everything through it in order to have the features expected on a non-smartwatch.

I guess devs can argue that once persistent storage will be operative, people would only need to pair the device once and then time and date can be stored on the device, and there wouldn't be necessary to change the time when traveling or on DST, and all those reasons are logical and will make user's life easier, but a simple manual setting would be cool if that goal is far away, implementation is not difficult and internal storage has space enough.

schultzryan, 2021-09-23:
Would the alarm feature recently merged be a good jumping off point for being able to set the time locally instead of requiring the device to be paired?

https://github.com/JF002/InfiniTime/pull/662#issuecomment-926118024

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/364

closed 2021-10-02 by Avamander