InfiniTimeOrg/InfiniTime #689: wake on wrist sleep mode

Izaic, 2021-09-26:
It would be handy to have a way to set a range of hours where wake on lift will not turn on the watch. For example while sleeping youll be tossimg and turning and it can drain the battery a lot quicker.

At least, this would be a decent workaround for it being so sensitive.

Bugsbane, 2021-09-30:
The concept of having a sleep mode is useful beyond just saving power on screen wake up. This could also temporarily disable things like the step counter and put the watch into a lower power mode.

Discoverability could be enhanced by both having a short message onscreen when turning it on, and by having it also be disabled with a single press of the physical button when it's engaged.

kieranc, 2021-10-01:
I agree, it's on my list of things to try implement. I was thinking about using the existing mute notifications system activated on a schedule. I'll look into if there are power savings to be had by disabling the step counter but I don't think it will be significant. We can also potentially/optionally activate it automatically after a certain period of no movement.

Bugsbane, 2021-10-01:
The step counter was only an example. The point is that when in sleep mode, pretty much everything except listening for the physical button (do disable sleep mode) and alarms could be disabled to save power.

For my use cases at least, the ability to set a schedule would be a plus, but far less important than the ability to easily turn it off/ on manually. If both are possible together though then that's the best off both worlds... but the most work for devs! 😉

Personally I would love to see the flashlight app moved to the apps screen, and a "sleep" toggle put in its place on the settings screen.

kieranc, 2021-10-02:
I've given this a bit of thought and I think the sleep mode currently ought to disable wake modes and bluetooth (disconnect and disable advertising). I don't think it is feasible to power off the hardware, same with the accelerometer, but if they're not in use the idle current is very low. Both bluetooth and accel may be useful during sleep for sleep tracking but the ability to stop the device waking up while you sleep would certainly be useful. This could be in addition to a quiet hours option which either just disables notifications, or enables sleep mode, user selectable? Since some people may want quiet hours but still be awake/want to check the time.

dozadoesit, 2022-04-01:
Also I just would like to add that there is another request for sleep tracking. This could be an easy solution.... when sleep mode is enabled we could consider the person is sleeping and then do "restless" tracking for sleep quality rather than rely on the heart rate monitor. This is similar to how fitbit used to do sleep tracking in it's earlier models should be low power as it only relies on the accelerometer.

closed 2022-08-21 by Riksu9000

Bugsbane, 2023-12-09:
I'm not sure how accurate the heart rate monitor is at this point as it was pretty bad but keeps getting better and better. That said if it's reasonably accurate, my preference would still be to use it, but keep the battery draw down by not using it continuously. If we only do a reading once every 15 minutes, I don't imagine it would be a huge power draw, but it would add significantly to the accuracy of the sleep tracking.

PR #1261 (2022-08-21): Add sleep mode
A sleep mode is needed so the watch can be worn during the night without causing the menus to activate.

Sleep mode disables notifications, touch wakeup, motion wakeup and chimes.

[image]

Fixes #689