InfiniTimeOrg/InfiniTime #690: infintime  1.5.0 lose time if reset when heart rate is enabled

lman0, 2021-09-26:
hi , 

today after installing 1.50 , and validated the firmware.
then i tried to reset in order to check if the time was up after reset 
and it was (that's a good point!! )

then after i decided to launch the heart rate.
and do a reset again 
and then it lost the time 

so the the bug found

ajack2001my, 2021-09-27:
@lman0, unable to reproduce your reported issue. I started the HR app, then pressed the back button to do a reset and the time still displayed correctly. Am I doing something wrong?

lman0, 2021-09-29:
@ajack2001my in the second reset , infinitme have been  disconnected from phone after time was retrieved 
and phone bluetooth was disabled  also as well.
Otherwise , the smartwatch would have picked instantly the phone , and updated the time on spot

JF002, 2021-09-30:
I cannot reproduce this issue either. Could you try to shoot a video showing exactly how to reproduce?

lman0, 2021-10-04:
hi,

i can't no longer test this since my pinetime died . maybe it was a hardware malfunction.
the replacement  pinetime i have (i had buy the 3 pinetime batch) , i can't  reproduce it .
i'm closing it then

closed 2021-10-04 by lman0