InfiniTimeOrg/InfiniTime #696: Timer/Stopwatch/Alarm apps should be consolidated

calhix, 2021-09-28:
It may be just me, but I think it would be nice for them all to be under one app icon (I prefer the icon of the alarm) and then there be 3 buttons for the timer, stopwatch and alarm, possibly with the same UI design as the buttons under settings (where Display, Wake Up, etc. are located). At the very least, I feel that the three apps should be grouped together.

Izaic, 2021-09-28:
I think having them grouped near each other in the app list makes sense

Riksu9000, 2021-09-28:
Tabs would take up a lot of precious screen space. They wouldn't even fit with the current UI, and it would probably end up being too cramped if it was changed. Swiping would also need indication that there is something there.

It might make sense to move them together, however everyone is used to the apps being in certain places, so I'm not sure it's a good idea to just move them around.

This was also suggested in #366 comments

calhix, 2021-09-28:
I'm not sure if you understood me correctly or I just explained poorly, I'm not talking about adding tabs in the app list so screen space is irrelevant, this would improve screen space since I'm talking about combining the three apps under one app icon. You would just click the icon and the next screen would have the three tabs called Alarms Timer and Stopwatch with a similar UI as the settings page. But even if that's not the best way forward I honestly don't get the excuse of not going ahead with moving a few apps around when it would make much more sense in the end and at most maybe just take a day or so to get used to it. It would be much more logical to keep similar apps grouped together.

Riksu9000, 2021-09-28:
I guess I didn't understand it correctly at first. So it would be like an app folder? Not sure if it's better to add more "layers" to navigating the interface. It would also have to be hard coded for now at least, so the "folders" couldn't be changed by users.

If we reorganized the apps now, maybe it isn't a big deal, but what about when a new app is introduced? Then we might have to shuffle everything around again, and it's definitely not a good idea to keep shuffling the apps every few releases.

escoand, 2021-09-28:
> I guess I didn't understand it correctly at first. So it would be like an app folder? Not sure if it's better to add more "layers" to navigating the interface. It would also have to be hard coded for now at least, so the "folders" couldn't be changed by users.

+1

> If we reorganized the apps now, maybe it isn't a big deal, but what about when a new app is introduced? Then we might have to shuffle everything around again, and it's definitely not a good idea to keep shuffling the apps every few releases.

I do understand this point, but this is only a problem when using this grid layout. For me the current layout wastes many space which results in tiny pixelated icons. Additionally the icons are not self-explanatory to me (why is the clock symbol the alarm app?). A solution could be an alphabetical ordered list like the settings.

But honestly I do more like the larger quick settings menu (dont know how it's called) and would ignore the shuffelling "problem". Having only 2 icons in a row is also shuffeling half often than 3... (when just counting the position within a row) ;-)
For me PineTimeLite by Joaquim is a much nicer app overview https://github.com/joaquimorg/PinetimeLite/blob/developLite/images/navigation.png

calhix, 2021-09-28:
> I guess I didn't understand it correctly at first. So it would be like an app folder? Not sure if it's better to add more "layers" to navigating the interface. It would also have to be hard coded for now at least, so the "folders" couldn't be changed by users.

Yeah so it may not be the best idea, I don't know what it is but to me the app section just seems convoluted and not perfectly clear what the function of each app is. I love the settings menu though. 
> 
> If we reorganized the apps now, maybe it isn't a big deal, but what about when a new app is introduced? Then we might have to shuffle everything around again, and it's definitely not a good idea to keep shuffling the apps every few releases.

This is true and I agree for the most part I just think these 3 apps are about the most basic functions imaginable for a watch and would most make sense being the first three apps in the list and together.

calhix, 2021-09-28:
> But honestly I do more like the larger quick settings menu (dont know how it's called) and would ignore the shuffelling "problem". Having only 2 icons in a row is also shuffeling half often than [...]

Oh wow this looks really good.

PR #1169 (2022-06-06): Move basic apps to the top of the app list
Moved timer and alarm next to the stopwatch on the first row. Music was moved to where timer was and navigation was moved to where alarm was.

[image]
[image]

Closes #696
closed 2022-06-06 by JF002