InfiniTimeOrg/InfiniTime #699: waving arms counts steps

watchdominiondotcom, 2021-09-28:
hi basically what the title says you move your arms around and it counts steps.

tmilburn, 2021-09-29:
The step counter functionality uses the in built step counter algorithm of the BMA423 accelerometer chip. Unfortunately some motion is mistaken for footsteps. All smart watches give some false positives but I haven't seen that the pinetime is significantly worse than other smart watches.

watchdominiondotcom, 2021-09-29:
i am aware of that but it seems to better on the ice apple. i saw video comparing lots of watches and pinetime on steps. i aware it is improving. i saw it adds ten steps to one arm wave. and lags on adding the true steps. like you said and i am. we know it happens. 

> The step counter functionality uses the in built step counter algorithm of the BMA423 accelerometer chip. Unfortunately some motion is mistaken for footsteps. All smart watches give some false positives but I haven't seen that the pinetime is significantly worse than other smart watches.

tmilburn, 2021-09-30:
It deliberately lags on adding steps so that it can improve the accuracy of step counting. The inbuilt step detector (as opposed to step counter) detects steps in real-time but is less accurate and isn't currently used by pinetime.

watchdominiondotcom, 2021-09-30:
right the thing is it still counts steps washing dishes sometimes on desk etc. the lag could be fine. just the extra steps is and issue.

Avamander, 2021-09-30:
Too bad, what you're asking is simply not possible to do on a wrist.

JF002, 2021-09-30:
The step counting is done in the hardware (the BMA42x sensor). It works by processing the X/Y/Z acceleration from the sensor and trying to interpret those values as a step. Also, the signal processing is probably a bit simple to keep the device operating in low power.
It's indeed really easy to make a move that would be interpreted as a step. Simply move the watch up and down 10 times and hop, it thinks you walked 10 steps! That's actually how I tested the implementation with my dev unit on my desk !

Can we do anything about it? Maybe : 
 - We can re-implement a more powerful algo in software. We would have the control on the signal processing, but it'll use more cpu/battery power.
 - We can also check the BMA42x doc for some settings on the step counting processor. It might be possible to fine tune some parameters of the algorithm to make it more or less sensitive to motion.

But in the end, remember that the sensor in mounted into the watch, and the watch is attached to your wrist, which is probably the part of the human body that is most in motion. Which means that the acceleration signal is very noisy and difficult to process accurately.

Also, as I have already said in other similar questions : I consider this 'step counting' measurement as a quantity of moves I've done during the day : it's higher when I work around the house and when I do groceries than when I just sit behind my desk for the whole day. It's probably not showing an accurate number of steps, but it's more or less proportional with to the "physical motions" I did with my body. To be honest, I have already thought about renaming the 'step count' into 'energy points' :-)

I'm not saying it's impossible to improve the accuracy, or that we won't do anything to improve it, but I think it won't be easy ;)

Hensch-S, 2021-09-30:
I thought that too. Its an indicator of movement. While doing office work the whole day it counts around 1500 but the same time on a building site will end up with around 6000. So this doesn't equal the steps I've done bit I know it was more movement and healthier for me ;)

kieranc, 2021-10-01:
I think we've seen several times that the step count produced by the PT is very similar (within 1% or less!) to that produced by other major smartwatches/fitness bands, and IMO if it's good enough for them it's good enough for us. There are certainly more pressing issues, the step count is not meant to be a 100% accurate count of the number of step you've made in a day but a general idea of your activity level.

Avamander, 2021-10-02:
I'll consider this a wontfix, unless someone comes and implements some other algorithm.

closed 2021-10-02 by Avamander

rcuddy, 2021-10-14:
I have been experimenting with steps myself and have the same problem but from a different perspective. Because it's seems to only be tracking based on arm swing it feels like you get no steps counted if you're not actually swinging your arms, ie hands in pockets, actually looking at the watch, carrying a coffee cup level in that hand, etc. I need to test a bit more but it feels like I'm losing a lot of step tracking to the fact that I'm often  not moving with much arm sway and it's not responding to gravity aligned bounces like I get counted when using other trackers

 I haven't looked into the "step counter algorithm of the BMA423 accelerometer" to see what it's actually doing but would it be worthwhile trying something like https://www.aosabook.org/en/500L/a-pedometer-in-the-real-world.html (which I was looking at for a different wearable project) as a software calculation using the chips XYZ values instead of asking it what it thinks a step is? I might try this myself but I've never done any embedded development so this might take me a while to get something useful