InfiniTimeOrg/InfiniTime #702: [Feature Request] Several improvement ideas for torch watchapp

adocampo, 2021-09-29:
I use a lot the torchlight of my watch. I have a baby and when he's grumbling at nights, I always tiptoe in quietly and in the dark. I do not want to turn on the lamps if he's not really crying, or if I don't need to change his diaper, so this app is essential for me, and I was very glad when I saw it by default on InfiniTime!

Sometimes, though, I have to hold him with one arm while with the other I give him a drink, pick up a thermometer, or anything else.
At those times, I may want to turn off the torchlight, so the baby doesn't wake up or get distracted by the light and fall asleep quickly.
Interacting with the torch while holding a baby, trying to screw something on, or performing almost any task that requires the use of one hand (not to mention those who are one-handed), becomes a real challenge, besides it turns on or off accidentally quite often.

 - I would strongly suggest switching it off and on instead of tapping, by shaking the wrist (not to be confused with lifting the wrist, but to shake to one side to the other in rapid succession).
 - I would also remove the torchlight icon completely from the screen, especially when it is turned off as it lights quite a lot in complete darkness, and yet it can let the user know the app is running, it is quite easy to press the back button and exit the torchlight watchapp.
 - Finally, the swipe gestures could be used to increase or decrease brightness when turned on.

I said several times now, and I don't want to be so repetitive, I came from pebble, and I tested some torchlight watchapps, and the best one, IMHO, was one which you simply needed to shake the wrist to turn it off and on again.

JF002, 2021-09-30:
Thanks for your feedback based on actual use-cases! I was wondering how users were using the torchlight and I understand it can be really useful in some cases ;)

adocampo, 2021-09-30:
For me is an essential watchapp... in pebble you were able to configure all of its 4 physical buttons to start an app, I had configured one of them to start the torchlight. That's how much important it's to me. If we were being able to configure our only physical button on PT, I would probably set the torchlight as well, well, I would doubt between the torchlight or a possible 2FA (#709) 😛

Bugsbane, 2021-10-01:
I use the torch quite a lot, too. Mainly I'm using it at night when I have most of the lights off and I'm looking for something. I don't want to switch the main house lights on as too much light at night interrupts melatonin release and lowers sleep quality,  so I love having a fairly dim light always available on my wrist.

That said my use case is a bit different. I usually have both hands available. I'm more interested in not waking myself up,  so for me, vigorous shaking of my arm isn't helpful. That said, I'm not against *also* having this method to turn the touch on, as long as it isn't accidentally triggering in the night when I'm asleep. My ideal for my personal Ueda though would probably be double pressing the physical button to activate the flashlight as it's quick and doesn't require either vigorous movement or navigating menus.

adocampo, 2021-10-03:
Well, I wouldn't say the movement of the wrist needs to be 'vigorous'. Just twist the wrist to one side, then to the contrary quickly. If you are used to wearing a metal watch band, the movement is the same, just a little more energetic.

Anyway, I would also love to have quick actions with the physical button, it would be great to double press the physical button to start it (I don't even know if that is possible, but having quick actions in double and triple press would be great, to start at least two apps)